react-native-camera/react-native-camera #2: Recommendation

mrjjwright, 2015-04-01:
For what it's worth, I think this is the best camera library out there to wrap.  Just my opinion, using it in my app and it's awesome.

https://github.com/IFTTT/FastttCamera.

lwansbrough, 2015-04-01:
Looks nice, though I'm wondering if it's a good idea to wrap third party libraries. I think it would be nice if this library didn't have dependencies on other iOS libraries - I'm open to have a discussion about it.

mrjjwright, 2015-04-01:
Totally up to you.  I know I won’t have a problem with it because there are a lot of very high quality third party libs out there and it saves a ton of work.  There will be lots of camera modules and it’s cool you are leading the way.  Perhaps you will find some good code in there.

lwansbrough, 2015-04-01:
Yeah, for now I think it's probably best to stick to core libs since RN is so young. Once adoption picks up I'm sure someone will fork this to add a third party library of some sort - not sure I want to make that my concern right now. :)

mrjjwright, 2015-04-01:
I personally don’t think the decision has anything to with the age of React.  It’s more the trajectory and stability of the third party lib you are wrapping versus writing one from scratch.   In a lot of cases I trust the battle tested nature of a third party lib, especially one that tis used in several production apps.

lwansbrough, 2015-04-01:
I suppose I have no real reservations about using a third party library, but I'd like to hear what other people think too. I'll leave this open for now - I'd like to hear from some more people before making any decisions like that. Also I'm not an Objective C developer so there's something to be said about using a hardened 3rd party lib. ;)

mrjjwright, 2015-04-01:
Absolutely cool.  Not suggesting anything for this lib at all.

lwansbrough, 2015-04-24:
I'm going to close this. The camera is working well without the suggested library, and we currently do not have a way to incorporate pods into React Native modules in a nice way.

closed 2015-04-24 by lwansbrough

mrjjwright, 2015-04-27:
Oh sorry, I should of closed that.  Thanks!