react-native-camera/react-native-camera #5: Build Fail

voronianski, 2015-04-01:
I've got a build fail everytime with such traceback:

```bash
RCTJSLog> "Error: 
 stack: 
  componentWillMount                      index.ios.bundle:38400
  chainedFunction                         index.ios.bundle:3587
  mountComponent                          index.ios.bundle:15901
  ReactCompositeComponent_mountComponent  index.ios.bundle:4847
  mountComponent                          index.ios.bundle:4916
  mountChildren                           index.ios.bundle:20393
  initializeChildren                      index.ios.bundle:19890
  mountComponent                          index.ios.bundle:20044
  mountComponent                          index.ios.bundle:4916
  mountComponent                          index.ios.bundle:15919
  ReactCompositeComponent_mountComponent  index.ios.bundle:4847
  mountComponent                          index.ios.bundle:4916
  mountComponent                          index.ios.bundle:15919
  ReactCompositeComponent_mountComponent  index.ios.bundle:4847
  mountComponent                          index.ios.bundle:4916
  mountComponentIntoNode                  index.ios.bundle:15382
  perform                                 index.ios.bundle:5979
  batchedMountComponentIntoNode           index.ios.bundle:15401
  perform                                 index.ios.bundle:5979
  batchedUpdates                          index.ios.bundle:13502
  batchedUpdates                          index.ios.bundle:4511
  renderComponent                         index.ios.bundle:15453
  ReactMount__renderNewRootComponent      index.ios.bundle:4847
  render                                  index.ios.bundle:1142
  renderApplication                       index.ios.bundle:33818
  run                                     index.ios.bundle:33764
  runApplication                          index.ios.bundle:33786
  jsCall                                  index.ios.bundle:7143
  _callFunction                           index.ios.bundle:7390
  applyWithGuard                          index.ios.bundle:877
  guardReturn                             index.ios.bundle:7192
  callFunctionReturnFlushedQueue          index.ios.bundle:7399
 URL: http://localhost:8081/index.ios.bundle
 line: 38400
 message: undefined is not an object (evaluating 'NativeModules.CameraManager.checkDeviceAuthorizationStatus')"
```

lwansbrough, 2015-04-01:
Working on a fix for this now, sorry I didn't realize this problem since I'm working on a project which includes the files directly.

lwansbrough, 2015-04-01:
Hey, please see the updated getting started section. Also make sure that the module is up to date.

closed 2015-04-01 by lwansbrough