react-native-camera/react-native-camera #10: `takePicture` captures first frame from output buffer instead of current frame

lwansbrough, 2015-04-03:
When you take a picture, the picture taken isn't actually the picture in your frame when the image was supposed to be captured. Instead, the first (at least, I think it's the first..) frame that is outputted to the `AVCaptureStillImageOutput` instance is the frame that is captured by `captureStillImageAsynchronouslyFromConnection`. I'm looking for some help on this, as I'm not really sure what the problem is.

kocyigityunus, 2015-04-07:
i think this is related to  : https://github.com/facebook/react-native/issues/725

you need to create an array as singleton. add all objects on manager's init method to that. and on a callback or some time later check if an object doesn't have a reactTag. if object doesn't have a reactTag it's a temp object for react and will not be on the js side. 

for reactTag look at "UIView+React.h"

lwansbrough, 2015-04-08:
@kocyigityunus Sorry, I'm not quite sure I'm seeing the correlation here. Do you think you can elaborate on how this affects the `captureStillImageAsynchronouslyFromConnection` callback?

kocyigityunus, 2015-04-08:
i think there is 2 layer when started. and only one of them continuing to update after first round. and ypu are trying to take picture from the first one.

lwansbrough, 2015-04-08:
That definitely sounds possible. But my callback doesn't reference a view (which, as per your instruction, I would use to check for `reactTag`), so I'm not sure how I could make sure the callback is referencing the right objects. Would I want to somehow associate each object with the current camera view's `reactTag` during the init?

lwansbrough, 2015-04-08:
@kocyigityunus Maybe the best solution is to store the objects inside the camera view?

kocyigityunus, 2015-04-08:
actually. on the init method at the manager. reactTag is always null. after first round or after some method call the reactTag fill with the right id to one of two objects. so i think you can continue to manager as a singleton manager. but you need to seperate that two views. 

i am still trying to learn this manager things etc.

lwansbrough, 2015-04-08:
Yeah I'm new to this too. And Objective-C programming hahah. I'm going to try and store all state in RCTCamera, and then delegate manager method invocations to the RCTCamera. This should keep the manager stateless (which I think is preferred?)

lwansbrough, 2015-04-08:
That worked!

kocyigityunus, 2015-04-08:
congratulations. i think this will be a great reference project for beginners.

closed 2015-04-08 by lwansbrough