react-native-camera/react-native-camera #13: Viewfinder does not work

brianchirls, 2015-04-16:
Adding this as noted under known issues in the readme, so we can discuss possible solutions.

It looks like you're not adding the `ViewfinderView` as a subview on `RCTCamera` anywhere. React does not take care of this automatically. I added some debug code to `layoutSubviews` and saw that the viewfinder does not have a `superview`, nor does it show up in the list of subviews, which is empty anyway.

However, that's not enough to solve the problem. I noticed in the old code that uses an earlier version of react-native, the viewfinder only works some of the time. It seems to fail whenever the app takes a little longer to start up, especially if I delay it by setting breakpoints in the debugger. When I reload the JS part of the app, it starts up faster and the preview works fine again. So perhaps it's a timing issue?

I did find a few claims that the preview layer needs to be set up on the [main thread](http://stackoverflow.com/questions/16515921/get-camera-preview-to-avcapturevideopreviewlayer). Maybe that will help.

lwansbrough, 2015-04-16:
Re: the viewfinder not being added as a subview - I'll take a look at this first, obviously sounds like it would be the major issue.

lwansbrough, 2015-04-16:
As for the camera not starting up sometimes, I think I have fixed that by moving most of the logic out of the RCTCamera class itself and into RCTCameraManager. I hope that fixes it anyway..

brianchirls, 2015-04-16:
It looks like you moved everything out of the manager into the camera and then back again. What prompted that decision?

lwansbrough, 2015-04-16:
@brianchirls I initially moved everything into the view in the hopes that I would be able to instantiate multiple different cameras at once, but I quickly realized this was the wrong choice for two reasons: You can't run multiple cameras at once, and React Native creates a shadow view which ended up having the access to the camera in some cases and so the viewfinder wasn't working for the "real" (non-shadow) view. I also think it's much cleaner to have the manager managing the view.

brianchirls, 2015-04-16:
Okay, that makes sense. Thanks.

lwansbrough, 2015-04-17:
Please see the latest release. The viewfinder issues *should* be resolved!

closed 2015-04-17 by lwansbrough