react-native-camera/react-native-camera #14: Compatibility with react-native 0.3.11 or above !

developerworks, 2015-04-21:
I created a new project with:

```
react-native init ReactNativeCameraTest
```

It can be running on Simulator and iPhone4S

And i install `react-native-camera` with:

```
npm install react-native-camera@latest --save
```

and follow README introduction, add the RCTCamera project to ReactNativeCameraTest, It shows the error:
https://github.com/facebook/react-native/issues/917

And shows a white screen on my 4S.

My Env:

XCode 6.2
Node: iojs 1.7.1
React Native: 0.3.11

lwansbrough, 2015-04-21:
Looks like this might be called by the `react-native` dependency. This has already been patched: https://github.com/lwansbrough/react-native-camera/blob/master/package.json but I haven't published those changes to npm. I suppose I should do that if the module is breaking stuff.

lwansbrough, 2015-04-21:
@developerworks I just pushed v0.1.1 to npm - let me know if that fixes it.

developerworks, 2015-04-21:
@lwansbrough 

It's fixed.

closed 2015-04-21 by developerworks