react-native-camera/react-native-camera #15: Add startScanning and stopScanning callback method.

developerworks, 2015-04-21:
There is a project which added `startScanning` and `stopScanning` based on your project.

https://github.com/lifuzu/ReactNativeBarcodeScanner/blob/master/iOS/RCTBarcodeScannerManager.m#L168

But that project is not a module project , It's a full xcode project and can not be added as Library dependency

lwansbrough, 2015-04-21:
The problem with adding the `startScanning` and `stopScanning` methods is that it ends the camera session - which in turn stops the viewfinder from updating. I decided instead to create an event handler prop that you can include which will receive data whenever a barcode is in view. Have you tried this out? Right now there's no rate limiter so while the barcode is in view, the event handler gets spammed.

steffenmllr, 2015-04-21:
I was also working on something like this, which kind of already works. The only problem I have is that the video sometimes freezes. Here is what I have so far: https://gist.github.com/steffenmllr/2eca6921525e3b9c7940

I have about 1 week in objective-c experience so not the best code I've written in my live :)

lwansbrough, 2015-04-21:
Thanks for your input guys. As far as I'm concerned, the barcode functionality is a solved problem. It needs to be tuned, but I don't necessarily agree that the API needs to change from it's current form. If you have a concern regarding the implementation, please don't hesitate to bring it up here.

lazaronixon, 2015-04-23:
I did this component using react-native-camera to simplify barcode scan... working 100% and easy..
Thanks a lot @lwansbrough you are the guy.
Some suggestion or enhancement is welcome.
https://github.com/lazaronixon/react-native-qrcode-reader

lwansbrough, 2015-04-23:
@lazaronixon Thanks for sharing :)

closed 2015-04-23 by lwansbrough