react-native-camera/react-native-camera #16: Image Manipulation

rt2zz, 2015-04-22:
How might image manipulation be supported? Namely scaling and cropping. Right now I have patched the module to crop the image square:
```
            // Crop logic
            CGImageRef imageRef = CGImageCreateWithImageInRect([rotatedImage CGImage], clippedRect);
            UIImage * croppedImage = [UIImage imageWithCGImage:imageRef];
            CGImageRelease(imageRef);
            NSString *imageBase64 = [UIImageJPEGRepresentation(rotatedImage, 1.0) base64EncodedStringWithOptions:0];
            NSString *croppedImageBase64 = [UIImageJPEGRepresentation(croppedImage, 1.0) base64EncodedStringWithOptions:0];
```
It would be nice to have an API to do arbitrary cropping/resizing, but I am not sure how to go about implementing this (iOS newb). Perhaps a second method called `takePictureWithManipulations` which takes a configuration object. The API could emulate the canvas api `ctx.drawImage(image, sx, sy, sWidth, sHeight, dx, dy, dWidth, dHeight)`

Another possibility, although this is something I know nothing about: we could return an image handle which can then be consumed to other image manipulation API's, this way we are not overburdening react-native-camera.

Thoughts?

lwansbrough, 2015-04-23:
Personally I'd prefer to keep image manipulation separate from the camera module, otherwise we'd be opening up a whole other can of worms. It should be pretty easy to pass the captured image into an image manipulation module though. Just take the output from `takePicture` and pass it to another module, for example:

```javascript
takePicture(err, data) {
  ImageManipulationManager.loadFromData(data);
}
```

rt2zz, 2015-04-23:
Makes sense, although I think it is expensive to cross the bridge 3 times
with a large payload, so it would be nice to have the ability to get a
string image id rather than the full base64 image. Not sure how the image
would be stored/passed around on the iOS side though...

lwansbrough, 2015-04-23:
@rt2zz I agree, perhaps a singleton could be created which could be accessed by any module.

rt2zz, 2015-04-26:
I am playing with some possible implementations in a fork. Right now I am writing the image to disk, which while slower than keeping it in memory, is nice because I can pass the uri directly to an Image component. Also it will likely play nicely with other modules like https://github.com/kamilkp/react-native-file-transfer. In my very rough testing this reduces the time it takes to take a picture and render a preview image down from ~1s to .3s.

Would you consider adding a method such as `capturePictureToDisk` which calls back with the local image path?

If you are interested and we can figure out an agreeable api I will make a PR.

lwansbrough, 2015-04-26:
@rt2zz That's fantastic, I'd happily accept a PR for such improvements. `capturePictureToDisk` sounds reasonable, the other option could be to add an optional object before the callback on `takePicture` (doesn't require an API change - just a type check) which would have a propertly like `target: 'disk'` or something. Use your discretion - I think I prefer the latter though as it prevents the API from becoming cluttered.

josebalius, 2015-04-29:
@rt2zz @lwansbrough this is a very similar conversation to https://github.com/facebook/react-native/issues/575, would love to see @rt2zz solution i think it is exactly what we need.

rt2zz, 2015-04-29:
@josebalius I need to clean things up and make a proper pull request, right now it always saves to the same filename "temp". See sample code below.

RCTCameraManager.m additions:
```
RCT_EXPORT_METHOD(takePictureToDisk:(RCTResponseSenderBlock)callback) {
    [[self.stillImageOutput connectionWithMediaType:AVMediaTypeVideo] setVideoOrientation:self.previewLayer.connection.videoOrientation];
    [self.stillImageOutput captureStillImageAsynchronouslyFromConnection:[self.stillImageOutput connectionWithMediaType:AVMediaTypeVideo] completionHandler:^(CMSampleBufferRef imageDataSampleBuffer, NSError *error) {

        if (imageDataSampleBuffer)
        {
            NSData *imageData = [AVCaptureStillImageOutput jpegStillImageNSDataRepresentation:imageDataSampleBuffer];
            UIImage *image = [UIImage imageWithData:imageData];
            UIImage *rotatedImage = [image resizedImage:CGSizeMake(image.size.width, image.size.height) interpolationQuality:kCGInterpolationDefault];
            
            //save in documents directory
            NSString *tempName = @"temp";
            NSString *fullPath = [self saveImage:rotatedImage withName:tempName];

            callback(@[[NSNull null], fullPath]);
        }
        else {
            callback(@[RCTMakeError(error.description, nil, nil)]);
        }
    }];
}

- (NSString *)saveImage:(UIImage *)image withName:(NSString *)name {
  NSArray *paths = NSSearchPathForDirectoriesInDomains(NSDocumentDirectory, NSUserDomainMask, YES);
  NSString *documentsDirectory = [paths firstObject];

    NSData *data = UIImageJPEGRepresentation(image, 1.0);
    NSFileManager *fileManager = [NSFileManager defaultManager];
    NSString *fullPath = [documentsDirectory stringByAppendingPathComponent:name];
    NSLog(@"writing path %@", fullPath);

    [fileManager createFileAtPath:fullPath contents:data attributes:nil];
    return fullPath;
}
```
In RCTCameraManager.h additions:
`- (void)takePictureToDisk:(RCTResponseSenderBlock)callback;`

lwansbrough, 2015-04-30:
@rt2zz This looks great, might I suggest `NSString *uuid = [[NSUUID UUID] UUIDString];` for the file name?

swennemans, 2015-04-30:
@rt2zz awesome! Really looking forward to your pull request. The main reason for not adapting react-native for me was not be able to take a picture and upload it.

josebalius, 2015-04-30:
I agree with @lwansbrough just name the file with a new random UUID. However @rt2zz it looks great! Go ahead and submit that PR.

lwansbrough, 2015-04-30:
@rt2zz Don't worry about docs and stuff. I think I'm gonna refactor the API a bit after your pull request.

rt2zz, 2015-05-01:
For now I decided against adding `target` as an options parameter to takePicture because I am wary of having the same method return different data types... well both strings, but one a base64 image, the other a filepath.

That said, long term I imagine this module might support more methods like capturing video or capturing a still from the video preview feed - the most generalized and explicit semantics might be:
```
capturePicture(opts, cb)
captureVideo(opts, cb)
capturePreviewStill(opts, cb)
---vs---
capturePicture()
capturePictureToDisk()
captureVideo()
captureVideoToDisk()
capturePreviewStill()
```
As I type this, I am leaning toward the former.  `capture` is more descriptive than `take`, and 3 methods with options cleaner than 6+. In short, I think the API needs some work, but still not sure about the details, depending greatly on what features you intend to add long term.

Some other things to consider for the pull request ( #20 ):
* There is no automatic removal of old images, so a naive implementation could easily result in users accumlating large numbers of unwanted images on their devices.
* One way around this is to save the file to temp directory, which iOS will automatically cull. This has the downside of potentially losing images between app restarts, or after app crashes.
* A nice to have would be a configuration option for where to save the file. 
* Also need to test how this can interoperate with [CameraRoll](http://facebook.github.io/react-native/docs/cameraroll.html#content) which afaik does not accept data URIs nor local file paths
  * see [RCTImageLoader.m](https://github.com/facebook/react-native/blob/master/Libraries/Image/RCTImageLoader.m) - although I imagine core would be open to a PR to handle base64 / local files

rt2zz, 2015-05-01:
also 
1. I made a mini demo app for testing: https://github.com/rt2zz/reactNativeCameraDemo
2. With the front facing camera the image is previewed in "mirror" orientation, but the image is captured in normal orientation. Not sure if the intuitive handling is to return the mirrored image or normal image? Maybe this deserves an option as well.

lwansbrough, 2015-05-01:
@rt2zz 1.) Nice. 2.) I haven't done anything with this so I assume it's standard behaviour. It makes a lot of sense to me - we're much more used to seeing ourselves in a mirrored perspective, but of course when give someone the image it will look strange to them if it's still mirrored.

Regarding the method names, I agree that "capture" is probably a better verb. Perhaps these changes + a few others are enough to justify a major version bump? I don't really have a problem changing the API with a major version (especially considering most people will probably only have one or a few references to the methods in their app).

I was actually considering the possibility of just a single method `capture(opts, cb)` which would take an object like `{ mode: 'still', target: 'disk', flash: true }`. Thoughts?

rt2zz, 2015-05-01:
@lwansbrough agreed. I think having a single capture method is probably the cleanest API, and I also agree this api change requires a major version bump.

I have some ideas on adding capture to disk options, but I will likely not be able to implementing them for some time.

lwansbrough, 2015-05-01:
Alright, so you think I can merge that PR now and fix up the API for a v2 release? I'd also like to tackle the orientation problem for that release.

rt2zz, 2015-05-01:
Sounds good to me

closed 2015-05-01 by lwansbrough

lwansbrough, 2015-05-01:
Changes can be seen here: https://github.com/lwansbrough/react-native-camera/commit/6dfcfa134e3c5411cd35768efca29e4c14f1cac6 Thanks for your help @rt2zz

josebalius, 2015-05-01:
Beautiful work guys, thanks!

swennemans, 2015-05-04:
Awesome guys. @rt2zz I see a lot of cool modules you worked on. Thanks!

A question:
I have an use case where I want to take a picture and upload this to the server. What is the (theoretical) best approach?
 I could either use: `Camera.constants.CaptureTarget.memory` or `Camera.constants.CaptureTarget.disk`.  My initial though was to capture to memory and upload it using fetch. However, based on the `readme.md` that doesn't seem like the best option, because keeping it in memory negatively impacts response time. 

So based on this info, I would assume that the best option would be: using `Camera.constants.CaptureTarget.disk` and then  (for example) send the returned path URI to `react-native-file-transfer` (https://github.com/kamilkp/react-native-file-transfer). 

Correct? Or am I missing something important here?

Thanks!

lwansbrough, 2015-05-04:
Personally I'll be using react-native-file-transfer. ;)

swennemans, 2015-05-06:
@lwansbrough cool :)

Any experience already using react-native-file-transfer? I'm trying but cannot get it to play nicely together. Any tips?

I also opened an issue: https://github.com/kamilkp/react-native-file-transfer/issues/1

jtremback, 2015-05-09:
@rt2zz About saving the images to disk and cluttering the camera roll... Seems that for better or worse a lot of photo apps actually do this- Instagram for example. I think that maybe users are used to it.

archanasantoshwhatkar, 2018-01-02:
how to save camera clicked photos and videos in my project folder in react native?