react-native-camera/react-native-camera #17: Header Search Paths

romanonthego, 2015-04-27:
hey.
I've notice you've used same install pattern as many react-native modules with manual setting up Header Search Path for xcode project.
Problem will rise when node_modules under .gitignore (as it should) and some kind of CI is used - your module will crash builds.
more on this problem - https://github.com/corymsmith/react-native-icons/issues/15.
@corymsmith found it possible to just include it by default, so may be you could do it too?

lwansbrough, 2015-04-27:
This should be the case, I think it's there just as a sanity check. Have you experienced any of such problems?

romanonthego, 2015-04-28:
No, it's actually a case - if you are using CI and doing it right (aka - not including node_modules/ in repo).

Our app continued failing to build until react-native-icons was forked, Header Search Path set up and fork referenced as source in package.json. Until when Travis was using clean install of npm package without Paths set up and failing to find headers files.

lwansbrough, 2015-04-28:
@romanonthego So have you experienced build problems with this module? The Header Search Path is set in the project file.

lwansbrough, 2015-07-19:
Going to close this as I haven't received any new issues regarding the problem you've outlined, though I acknowledge without further investigation this may be an issue for some. I'll leave it to them to open an issue should that be the case.

closed 2015-07-19 by lwansbrough