react-native-camera/react-native-camera #23: Notes about base64 image & the CameraRoll

rt2zz, 2015-05-02:
To display the base64 image that is returned by this module:
```js
//style height & width and source.isStatic are all required
<Image
  source={{
      isStatic: true,
      uri: 'data:image/jpeg;base64,'+base64image,
    }}
  style={{height: 100, width:100}}
/>
```

Also images can be saved to the CameraRoll with the following patch: https://github.com/facebook/react-native/pull/1111

closed 2015-05-02 by rt2zz

jtremback, 2015-05-12:
Is putting the data over the bridge and a base64 conversion twice (once from the camera, once to the screen) a concern at all?

lwansbrough, 2015-05-12:
@jtremback Yeah it's slow and locks up other actions from taking place over the bridge while it is being sent as far as I know.

jtremback, 2015-05-12:
I found it quite slow as well. Is the base64 conversion option in this module ever a realistic option?

lwansbrough, 2015-05-13:
Potentially. It has mostly just been kept as the default for backwards compatibility and predictability.

jtremback, 2015-05-13:
Kind of feels like there needs to be some code around handling files and data "over the bridge" in React Native in general.

lwansbrough, 2015-05-13:
The bridge is not really designed to be that. React Native is designed to reduce the workload of building a native app, it's not really a "Javascript all the things!" solution. The current suggested method of using file URIs works just fine.

jtremback, 2015-05-13:
Off topic for this thread I guess, but I was more thinking about an
abstraction that would let you manipulate data that is on the objC side in
JS without actually copying the actual data over.

I guess saving it on disk and reading it when I need to is good enough for
now though.

lwansbrough, 2015-05-13:
I don't know how you think that would work - the only way to get data between JS and Objective-C is over the JS bridge.

ahh2131, 2015-08-26:
Sorry if this is a silly question, but how do I get the base64 data from the camera without saving the image to the camera roll?