react-native-camera/react-native-camera #24: Can only get aspect fit to work

jonsherrard, 2015-05-03:
Hi! First of all thank you for such a fantastic addition to the react-native eco system. It opens up so many interesting applications!

 - Regardless of the value I use with `aspect={Camera.constants.Aspect.fill/stretch/fit}` the aspect always seems to be fit
 - I've tried setting heights on the `<Camera />` element itself and also wrapping it in a `<View>` and setting height and width constraints on that.

 - Regardless I only ever get the vertical image with black lines at the side. (I have 60px of navbar at the top)

 - I've tried all the constants as strings as well, and also as capitalised strings which throws a deprecation invariant warning

- Any points would be very welcome. I'm using react-native-cli@1.3.0, and react-native@4.3.1, with camera@0.2.1

Thanks again for all your work on the react-native-camera element.

Jon

lwansbrough, 2015-05-04:
Looking into this now, thanks for reporting.

lwansbrough, 2015-05-04:
I can reproduce this - looks like I didn't catch the bug because the demo uses the screen's aspect ratio which matches the "fit" aspect. Hah.

lwansbrough, 2015-05-04:
Ah, oh boy. Don't think this fixed it correctly. Now everything is `fill`. 😁

closed 2015-05-04 by lwansbrough

jonsherrard, 2015-05-04:
Thanks Loch, I owe you a drink if you're even in London (UK)

lwansbrough, 2015-05-04:
Cheers. 🍻

commit bbce1ca7 (2015-05-04): Fixes #24
commit 631ea890 (2015-05-04): Actually fixes #24