react-native-camera/react-native-camera #26: Displaying the captured image.

antoinegrant, 2015-05-05:
So I'm to clear as to how we go about displaying the captured image from memory or from disk? Could give an example?

rt2zz, 2015-05-05:
To display the base64 image (i.e. target: Camera.constants.CaptureTarget.memory) the code is roughly
```js
<Image
  source={{
      isStatic: true,
      uri: 'data:image/jpeg;base64,'+base64image,
    }}
  style={{height: 100, width:100}}
/>
```
where base64image is the string returned by the capture method.

To display the `target: disk` image, the code is roughly
```js
<Image
  source={{uri: imageURI, isStatic:true}}
  style={{width: 100, height: 100}}
        />
```
where imageURI is the string returned by the capture method.

You will probably need to play with the width/height, and resizeMode prop to get everything the way you want it.

antoinegrant, 2015-05-05:
That very helpful, thanks.

closed 2015-05-05 by lwansbrough

Mayzie, 2017-05-25:
Hi there,

This does not seem to work any longer on React Native 0.44 for Android. How can I achieve this (displaying a temp, or disk, image)?

npomfret, 2017-05-25:
@Mayzie - what doesn't work?  do you have an error or a stack trace from logcat?

julienvincent, 2017-08-04:
Can confirm this no longer works. No problems on a simulator, but does not work on a real device

julienvincent, 2017-08-04:
No nevermind, after reinstalling it works as expected

pablodance, 2019-09-24:
brilliant - thanks very much @rt2zz  - saved me some head scratching..