react-native-camera/react-native-camera #28: Start camera on button click

flyingfeet, 2015-05-06:
Hey,

first of all I'm very new to react native and it's concepts. I'm evaluating cross-platform frameworks for my thesis.

I want to ask, if it's possible with your plugin to start the camera when a button ("take picture") was clicked, then of course take the picture and show the taken image below the button. I didn't get it working, just the background camera, but then I had problems with my TabBar which was in front but didn't work anymore...very freaky^^

Thanks and greetings from Germany!

lwansbrough, 2015-05-06:
I don't think your problem is really specific to this module, but perhaps I can help. The camera starts when its `<Camera/>` element is rendered. Thus, if you want the camera to start, render the `<Camera/>` element. If you don't want it rendered, you need a way to determine when it should be rendered and when it shouldn't. This is what state is used for in React.js/React Native. Based on what you've said, you want to render the camera and take the picture in the same instant. I don't think this is possible, as the camera must be rendered before you can call `capture` on its reference. Is there any reason why you couldn't have the camera running already, and then just capture the image when the "take picture" button is pressed? You can run the camera invisibly either by setting height/width = 0 or opacity = 0.

closed 2015-05-06 by lwansbrough

flyingfeet, 2015-05-06:
Thanks for your quick response!
I already thought about your solution at the end but then the user didn't know what picture he takes because he doesn't see any picture. Am I right?
My problem is also that I didn't get the camera in fullscreen mode in front of all other stuff. It's always in the background and blocks my TabBar navigation...