react-native-camera/react-native-camera #29: Write image to tmp directory instead of encoding it.

lukasreichart, 2015-05-06:
The capture function returns the image Base64 encoded. I'd propose to add another mode to the capture function:
Instead of returning the image to the callback, we write it to the <Application_Home>/tmp directory using a UUID as the filename. The callback just receives the UUID instread of the Base64 encoded data.

I could implement this feature easily but first wanted to check if anyone comes up with a better idea :D

rt2zz, 2015-05-06:
Agreed about having a configurable subfolder. It should probably default to a subfolder like "RCTCamera".

We did not use the tmp directory by default because it may be cleared out between app restarts, but in some applications this is probably desirable behaivor. 

Which leaves the question of desigining the options around this. Something like
`targetBaseDirectory` enum("documents", "tmp") - defaults to documents
`targetSubDirectory` string - defaults to "RCTCamera"

However I can see target options getting unwieldy quickly.

lukasreichart, 2015-05-06:
I think one option would be easier to use:
``` targetDirectory ``` defaulting to "documents/RCTCamera".
( If the directories used in ``` targetDirectory ``` don't exist we can easily create them. )

rt2zz, 2015-05-06:
I believe the only two valid base directories are `Documents` and `tmp`... and *Maybe* `Library`. So I would definitely want to enum that value. 

As a seperate question, perhaps targetOptions should be its own config object, much like source is an object in `Image`
```
targetOptions: {
  temp: (bool),
  subDirectory: (string),
}
```
Just throwing out ideas, I like to keep things explicit, but I know Loch likes to keep things tidy.

lwansbrough, 2015-05-06:
Just a thought - what if `target` could be `memory`, `disk`, or a directory? `disk` would basically just be a default directory like we're already doing it.

lukasreichart, 2015-05-11:
I think it makes more sense to add some options when using the target ```disk```, because when you define ```disk``` and ```target```, you have two ```captureTarget```'s with nearly the same functionality.
But in the @lwansbrough gets to decide, and I'd be really happy to start implementing this new feature :D

rt2zz, 2015-05-19:
After thinking about this more, and reading other peoples thoughts in various issues, I think there are several issues that need to addressed, perhaps with signifigant API changes:

1. Make default target be `disk`
2. Make default directory `/tmp`
3. Rename `memory` to `base64String`
4. Add `targetDirectory ` configuration option (like @lukasreichart originally proposed)

This I think will solve a few concerns. Saving to /tmp as a default simplifies the naive implementation. Renaming `memory` to `base64String` is both more explicit, and opens up the possibility for using `memory` to store the captured image in memory in some singleton, which may be a useful performance optimization for apps that need extremely low latency access and manipulation of the captured image.

As it appears this is going to be the defacto camera solution for react-native, now is probably a good time to open up the API discussion and refactor if necessary. In the meantime adding `targetDirectory` is probably the most obvious first step.

lwansbrough, 2015-05-19:
I'm in favour of doing whatever is best for the community, so let's decide what that is. What @lukasreichart and @rt2zz have proposed seems like a pretty solid solution. If anyone has objections or comments we'd like to hear them.

achang28, 2015-08-20:
So I may be wanting to look at this soon, except with a twist:  to see how I can build out a solution using Swift (because that's all I know -- to an extent, too).  Perhaps something that is invoked once the image is obtained.  And that "something" will/should provision the proposed solution above (which I like)...

rt2zz, 2016-02-17:
This issue is a bit dated. there is now a `target="temp"` option for both android and iOS thanks to @christopherdro :+1:

closed 2016-02-17 by rt2zz

christopherdro, 2016-02-17:
@rt2zz Just a heads up, I believe this applies only to photos. Temp directory for videos needs to still be added.