react-native-camera/react-native-camera #30: Uploading an image from disk, file not found.

swennemans, 2015-05-06:
Saving an image to disk works perfectly, huuuge performance boost versus in memory. :smile:  Also previewing the image works fine.
 
However, when trying to select the image, in my use case to cloudinary, the url doesn't seem to work. Is there something particular I have to do?
ATM: I save my image to CameraRoll with @rt2zz patch. And send the image with the assets URL. 
In my case the `assetsURL` works, the `fileURL` doesn't. In the case of cloudinary it will respond with `empty file`. 
Example of logged URLs:

`var assetURL = "assets-library://asset/asset.JPG?id=B37E1F30-C8F4-4648-8BF2-1B8BD638141D&ext=JPG”;`

`var fileURL = "file://var/mobile/Containers/Data/Application/A70929AB-C7B5-4DEE-8406-0EF4A5758C3F/Documents/3F60C854-5CE0-4A73-825E-97E0FD9092E1”;`

Comparing both URL the syntax looks different, is there any magic I have to do for the fileURL?

rt2zz, 2015-05-06:
are you using react-native-file-transfer ? 

upon very cursory inspection I think react-native-file-transfer has a bug  in supporting file:// tags

Above is a patch that I believe solves your use case, but I have not tested it

prashanthnani, 2015-11-25:
Please provide some code to save asset uri images to cloudinary through node.js using react native CameraRoll component