react-native-camera/react-native-camera #32: jsSchedulingOverhead (-5999ms) should be positive

jtremback, 2015-05-09:
Get this warning log when using .capture in development mode... I don't know that this is an issue, just FYI

lnut101, 2015-05-20:
Yes i also saw a lot of this warning, but with different values

bright-sea, 2015-05-27:
Yes, me too. happened when debug on devices. In index.ios.js, I have set a interval checking like:

<pre>
setInterval(FileSystemUtils.checkInboxFile, 3000);
</pre>

lwansbrough, 2015-07-11:
@jtremback Is this still happening?

pgmemk, 2015-07-12:
@lwansbrough Yes it is for me

jtremback, 2015-07-13:
Haven't seen this myself, for a while.

lwansbrough, 2015-07-14:
@pgmemk Are you using the latest version of the camera?

sunilp, 2015-07-19:
What it means is that the start date passed on the timer creation (Date.now() on JavaScript) is 5999ms ahead of the current time on iOS ([NSDate now]), when you're running the app on the device and the executor on your computer it's very unlikely that the times will be the same at the millisecond level. So no, don't worry about it.

can simply ignore.

lwansbrough, 2015-07-19:
@sunilp That's good news - is there a way we can suppress this notification?

closed 2015-07-21 by jtremback