react-native-camera/react-native-camera #35: orientationChanged sent to deallocated instance of RCTCamera

pmg103, 2015-05-19:
My app keeps crashing with this error:

`*** -[RCTCamera orientationChanged:]: message sent to deallocated instance 0x1246732e0`

lwansbrough, 2015-05-19:
Has the camera view been removed at this point? (ie. you've changed pages/scenes/states that makes the camera disappear)

pmg103, 2015-05-19:
No. Originally I was doing that, but I wondered if that might be the problem, so i switched to keeping the component on the page the whole time.

I'm handling `onBarCodeRead`. I'll post the code here tomorrow if you'd like.

lwansbrough, 2015-05-19:
Yeah sure, and I'll see if I can find some time to take a look at this tonight.

pmg103, 2015-05-20:
```javascript
  render() {
    var scanButton = this.state.isScanning ? (
        null
      ) : (
        <TouchableHighlight style={styles.button}
            onPress={this.onScanPressed.bind(this)}
            underlayColor='#99d9f4'>
          <Text style={styles.buttonText}>Scan Now</Text>
        </TouchableHighlight>
      );

    return (
      <View style={styles.container}>
        <Image source={require('image!logo')} style={styles.image}/>
        <Text style={styles.description}>
          Scan your barcode below
        </Text>
        {scanButton}
        <Camera
          ref="cam"
          style={styles.container}
          onBarCodeRead={this.onBarCodeRead.bind(this)}
          type={this.state.cameraType}/>
        <Text style={styles.description}>{this.state.message}</Text>
      </View>
    );
```

and

```javascript
  onBarCodeRead(e) {
    if (!this.state.isScanning) return;
    this.setState({
      isScanning: false,
      message: 'Submitting Code'
    });
    this.store.storeAndSend(e.data, function (response) {
      this.setState({ message: response });
    }.bind(this));
  }
```

Seems innocuous enough.

However, everything works fine for a few scans, but then I get the ```-[RCTCamera orientationChanged:]: message sent to deallocated instance``` error and the app crashes.

ortutay, 2015-05-23:
I am running into this issue as well. Here as simple component that reproduces the problem:

```
var Crash = React.createClass({
  getInitialState: function() {
    return {page: 'a'}
  },

  handleToggle: function() {
    if (this.state.page == 'a') {
      this.setState({page: 'b'});
    } else {
      this.setState({page: 'a'});
    }
  },

  render: function() {
    if (this.state.page == 'a') {
      return (
        <View style={styles.container}>
          <Text>hello</Text>
          <Button onPress={this.handleToggle}>Toggle</Button>
        </View>
      );
    } else {
      return (
        <Camera ref="cam" style={styles.container}>
          <Text style={styles.instructions}>Camera</Text>
          <Button onPress={this.handleToggle}>Toggle</Button>
        </Camera>
      );
    }
  }
});
```

To reproduce:
1) Run app on iPhone
2) Click "Toggle" (goes to camera)
3) Click "Toggle" again (goes to plain text page)
4) Rotate iPhone
5) Observe crash

pmg103, 2015-05-28:
You are adding an observer of NSNotificationCenter in `RCTCamera setOrientation()`:

```javascript
- (void)setOrientation:(NSInteger)orientation
{
    if (orientation == RCTCameraOrientationAuto) {
        [self.manager changeOrientation:[UIApplication sharedApplication].statusBarOrientation];
        [[NSNotificationCenter defaultCenter] addObserver:self  selector:@selector(orientationChanged:)    name:UIDeviceOrientationDidChangeNotification  object:nil];
    }
    else {
        [[NSNotificationCenter defaultCenter]removeObserver:self name:UIDeviceOrientationDidChangeNotification object:nil];
        [self.manager changeOrientation:orientation];
    }
}
```

 I think that maybe when the `RCTCamera` instance is deallocated you should remove the observer?

I was able to go ahead and use `react-native-camera`.in my project by actually just removing that observer as my app is fixed in portrait mode anyway.

commit 021e8b58 (2015-06-02): Fixes #35
closed 2015-06-02 by lwansbrough