react-native-camera/react-native-camera #37: Captured image does not crop image to  <Camera> viewfinder's ratio

ksokhan, 2015-05-23:
I expected the outputted image to have the same width/height ratio as the camera viewfinder. However, the returned image is always the full capture of the camera, irrespective of the camera viewfinder's dimensions.

lwansbrough, 2015-05-26:
This is intended - I could see this being a useful setting though.

ksokhan, 2015-05-29:
Ok thanks! I'll see if I can set up a PR for this.

closed 2015-05-29 by ksokhan

nickgerig, 2016-11-08:
This would be a great feature. I think it's also about user expectations and even privacy. A user will frame the photo based on the preview so any subsequent action on the actual photo needs to be re-evaluated by the user if it's different to the preview. eg. if you are going to upload a photo

Are the cropping rules the same on iOS and Android? It could be that the cropping of the actual photo is handled outside of this library if the preview cropping rules were consistent cross platform and documented.