react-native-camera/react-native-camera #39: I don't know why my phone6 to turn on the camera is too slow

soliury, 2015-05-26:
I just give a button on my first scence , and then , when click it can push a new scence which have the camera, but , after I click, it will take 3s to load, and the next , I try more time it go fast, but the first is too low.

rt2zz, 2015-05-26:
3s to load the scene or does the scene load immediately and it takes 3s for the camera preview feed to start?

soliury, 2015-05-26:
I change to a test component , it load quickly. So I think camera make the loading slow

lwansbrough, 2015-05-26:
Will need to see some code.

soliury, 2015-05-27:
This is my qrCode Component :

```
var React = require('react-native');
var Dimensions = require('Dimensions');

var Camera = require('react-native-camera');
var Icon = require('FAKIconImage');
var Button = require('react-native-button');

var { width, height } = Dimensions.get('window');

var {
    StyleSheet,
    View,
    Text,
    TouchableOpacity,
    Component,
    Navigator,
    VibrationIOS
    } = React;

var cameraSize = 250
var borderColor = 'rgba(255,255,255,0.6)'
var borderBoxSize = 35

var styles = StyleSheet.create({
    camera: {
        width: width,
        height: height,
        flexDirection: 'column', 
        justifyContent: 'center',
        alignItems: 'center'
    },
    header: {
        height: 80,
        width: 350,
        //backgroundColor: 'red',
        flexDirection: 'row',
        justifyContent: 'space-between',
        alignItems: 'center'
    },
    cameraView: {
        height: cameraSize,
        width: cameraSize,
        //backgroundColor: 'blue'
    },
    container: {
        height: 350
    },
    borderBox: {
        position: 'absolute',
        borderWidth: 2,
        height: borderBoxSize,
        width: borderBoxSize
    },
    borderLeftTop: {
        borderColor: 'transparent',
        borderLeftColor: borderColor,
        borderTopColor: borderColor,
        left: 0,
        top: 0
    },
    borderRightTop: {
        borderColor: 'transparent',
        borderRightColor: borderColor,
        borderTopColor: borderColor,
        right: 0,
        top: 0
    },
[... 335 lines cut ...]
                backdropBlur='dark'
                isVisible={this.state.isModalOpen}
                onClose={() => this.closeModal()}>

                <View style={styles.wrapper}>
                    <View style={styles.row}>
                        <Button
                            onPress={this._onLoginPress.bind(this)}
                            style={styles.button}>
                            <Icon
                                name='ion|camera'
                                size={28}
                                color='rgba(255,255,255,0.7)'
                                style={styles.icon}/>
                            <Text style={styles.buttonText}>扫码登陆</Text>
                        </Button>
                    </View>
                </View>
            </Modal>
        )
    }
}

module.exports = Login
```

soliury, 2015-05-27:
I think I have found the reason , it's the my phone problem , after I reboot my iphone, it works quickly.

closed 2015-05-27 by soliury

lwansbrough, 2015-05-27:
Interesting, thanks for letting me know. I'll keep an eye on it.

soliury, 2015-05-27:
But sometimes, it also slow....I don't know why

lwansbrough, 2015-05-27:
The QR code scanner spams the JS bridge if it is in the process of scanning, I've been meaning to fix that. I wonder if that has an effect on the speed of the app.

soliury, 2015-05-27:
Ok, I see. When you finish it , can you give me a message in here ? Now I am using this package in a project.

skizzo, 2018-10-25:
Still having this issue, 1 out of 3 times it takes about 4 seconds to initialize the camera component on my iPhone 6s..