react-native-camera/react-native-camera #40: Make it work with rem

jtremback, 2015-05-27:
I can help with this, but you'll need to put this on cocoapods first. Seems like the npm module and the cocoapods module should be owned by the same person.

https://github.com/ReactExtensionManager/rem#writing-modules-for-rem-for-module-authors

lwansbrough, 2015-06-02:
`pod spec lint react-native-camera.podspec`

```
 -> react-native-camera (0.2.8)
    - WARN  | Github repositories should use an `https` link.
    - ERROR | [iOS] Returned an unsuccessful exit code. You can use `--verbose` for more information.
    - NOTE  | [iOS]  react-native-camera/RCTCameraManager.h:1:9: fatal error: 'RCTViewManager.h' file not found
    - NOTE  | [iOS]  react-native-camera/RCTCamera.m:1:9: fatal error: 'RCTBridge.h' file not found
    - WARN  | [iOS]  react-native-camera/UIImage+Resize.m:136:22: warning: implicit conversion from enumeration type 'enum CGImageAlphaInfo' to different enumeration type 'CGBitmapInfo' (aka 'enum CGBitmapInfo') [-Wenum-conversion]
    - WARN  | [iOS]  react-native-camera/UIImage+Resize.m:170:13: warning: enumeration values 'UIImageOrientationUp' and 'UIImageOrientationUpMirrored' not handled in switch [-Wswitch]
    - WARN  | [iOS]  react-native-camera/UIImage+Resize.m:190:13: warning: 4 enumeration values not handled in switch: 'UIImageOrientationUp', 'UIImageOrientationDown', 'UIImageOrientationLeft'... [-Wswitch]
    - WARN  | [iOS]  react-native-camera/UIImage+Resize.m:18:17: warning: method definition for 'thumbnailImage:transparentBorder:cornerRadius:interpolationQuality:' not found [-Wincomplete-implementation]
    - NOTE  | [iOS]  react-native-camera/UIImage+Resize.h:12:1: note: method 'thumbnailImage:transparentBorder:cornerRadius:interpolationQuality:' declared here
    - WARN  | [iOS]  react-native-camera/UIImage+Resize.m:167:1: warning: category is implementing a method which will also be implemented by its primary class [-Wobjc-protocol-method-implementation]
    - NOTE  | [iOS]  react-native-camera/UIImage+Resize.m:15:1: note: method 'transformForOrientation:' declared here
```

lwansbrough, 2015-06-02:
Changed my podspec:

```
Pod::Spec.new do |s|

  s.name         = "react-native-camera"
  s.version      = "0.2.8"
  s.summary      = "A Camera component for React Native. Also reads barcodes."
  s.homepage     = "https://github.com/lwansbrough/react-native-camera"
  s.license      = { :type => "MIT", :file => "LICENSE" }
  s.author       = { "Lochlan Wansbrough" => "lochie@live.com" }
  s.platform     = :ios, "7.0"
  s.source       = { :git => "https://github.com/lwansbrough/react-native-camera.git", :tag => "v0.2.8" }
  s.source_files = "*.{h,m}"
  s.requires_arc = true
  s.dependency 'React'
end
```

And now I'm getting this:

`$ pod spec lint react-native-camera.podspec`

```
[!] /bin/bash -c camera
set -e
npm install

> ws@0.4.31 install /Users/loch/Library/Caches/CocoaPods/Pods/Release/React/0.4.4-17e98/node_modules/ws

> (node-gyp rebuild 2> builderror.log) || (exit 0)

  CXX(target) Release/obj.target/bufferutil/src/bufferutil.o

> contextify@0.1.14 install /Users/loch/Library/Caches/CocoaPods/Pods/Release/React/0.4.4-17e98/node_modules/jest-cli/node_modules/jsdom/node_modules/contextify

> node-gyp rebuild

  CXX(target) Release/obj.target/contextify/src/contextify.o

  SOLINK_MODULE(target) Release/contextify.node
npm WARN engine jest-cli@0.4.3: wanted: {"node":"0.8.x || 0.10.x"} (current: {"node":"2.2.1","npm":"2.11.0"})

npm WARN engine stacktrace-parser@0.1.1: wanted: {"node":"~0.10"} (current: {"node":"2.2.1","npm":"2.11.0"})

ld: library not found for -lgcc_s.10.5

clang: error: linker command failed with exit code 1 (use -v to see invocation)

make: *** [Release/contextify.node] Error 1

gyp ERR! build error 

gyp ERR! stack Error: `make` failed with exit code: 2

gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)

gyp ERR! stack     at emitTwo (events.js:87:13)

gyp ERR! stack     at ChildProcess.emit (events.js:172:7)

gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:199:12)

gyp ERR! System Darwin 14.0.0

gyp ERR! command "/usr/local/bin/iojs" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"

gyp ERR! cwd /Users/loch/Library/Caches/CocoaPods/Pods/Release/React/0.4.4-17e98/node_modules/jest-cli/node_modules/jsdom/node_modules/contextify

gyp ERR! node -v v2.2.1

gyp ERR! node-gyp -v v1.0.3

gyp ERR! not ok 

npm WARN prefer global eslint@0.9.2 should be installed with -g

npm ERR! Darwin 14.0.0

npm ERR! argv "/usr/local/bin/iojs" "/usr/local/bin/npm" "install"

npm ERR! node v2.2.1

npm ERR! npm  v2.11.0

npm ERR! code ELIFECYCLE

npm ERR! contextify@0.1.14 install: `node-gyp rebuild`

npm ERR! Exit status 1

npm ERR! 

npm ERR! Failed at the contextify@0.1.14 install script 'node-gyp rebuild'.

npm ERR! This is most likely a problem with the contextify package,

npm ERR! not with npm itself.

npm ERR! Tell the author that this fails on your system:

npm ERR!     node-gyp rebuild

npm ERR! You can get their info via:

npm ERR!     npm owner ls contextify

npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:

npm ERR!     /Users/loch/Library/Caches/CocoaPods/Pods/Release/React/0.4.4-17e98/npm-debug.log
```

commit 6f667f16 (2015-06-02): REM support (Fixes #40)
closed 2015-06-10 by jtremback