react-native-camera/react-native-camera #51: Not working if app initialization is triggered through location services

dizlexik, 2015-06-16:
I've implemented react-native-camera in my app and it's working amazingly, except for this one issue. My app uses location services to constantly scan for beacons. When a beacon is found in range iOS launches the app silently in the background so it can respond to the event. It then leaves the app running. When the app is started in this way the next time I open it (remember it's running in the background already) the camera does not work at all. I just see a black or white rectangle when I load up a Camera element.

I spent a little time digging around through the react-native-camera code to see if I could figure this out but my iOS dev skills are pretty lacking so I think I have to throw in the towel at this point :/  I don't know if something can be done to initialize the camera only after the app has become active on-screen or something, but I'd really love it if this weird edge case could be resolved because it's the only thing holding my app back from perfection :)

lwansbrough, 2015-06-21:
Interesting. I'll have to think about the best way to deal with this. It must be possible to detect if they app is running in the background and not run the typical camera initialization if it is.

jtremback, 2015-07-05:
The issue is that the camera is not being initialized properly?

lwansbrough, 2015-07-10:
@dizlexik I changed the way the camera is initialized. I'm not sure if the changes will fix your issue, but I'd appreciate it if you could give it a try. The latest version with the changes is not on npm so you'll have to install via the master branch if you want to try it out.

dizlexik, 2015-07-10:
It actually did fix the issue, great!

Although now I can't get bar codes to scan, so I can't use this yet :/  It looks like there was some work in that area but I don't see any documentation updates so I'm not sure if I need to update my code to get that working again. I don't have enough time to dig deeply but I took a quick look at one commit and tried adding `barCodeType={Camera.constants.BarCodeType.qr}` to my Camera component but onBarCodeRead still wasn't getting called.

Aaaaanyway, yeah whatever you did regarding initialization did fix my issue with location services. Thanks! I'm excited for the next release containing this fix and a working barcode scanner :)

lwansbrough, 2015-07-10:
Ok, closed a door and opened a window. Not too bad. Thanks for reporting that. I'll have a look at it.

closed 2015-07-10 by lwansbrough

lwansbrough, 2015-07-10:
So the `barCodeType={Camera.constants.BarCodeType.qr}` is unnecessary - those constants are only exposed so you can test the bar code scan `event.type` against them (if needed.) There was indeed an issue caused by the new initialization process. Fixed here: https://github.com/lwansbrough/react-native-camera/commit/5d306b54b8cb2670658061758ce427e9e1eecc4b