react-native-camera/react-native-camera #52: Camera start when app open, so iPhone will become hot

guodong000, 2015-06-18:


rt2zz, 2015-06-18:
You should use some conditional logic to only display the camera component when it is in use.

The Navigator component will keep the camera in the route stack, so you should either manipulate the stack or have some conditional logic in your camera scene e.g. {this.props.activeScene ? <Camera /> : null}

guodong000, 2015-06-19:
Well, I know...

lwansbrough, 2015-07-02:
I believe the capture sessions are started when the module is initialized, which would be the contributing factor to this issue. With the arrival of video capture I need to ensure these capture sessions do not start prematurely.

morukutsu, 2015-07-09:
I had the same issue and I added a function to enable the camera capture manually. 
See: https://github.com/morukutsu/react-native-camera/commit/db17ea9f2d03e1a5279efd2ff6d92702445f82b7

It obviously need more work but it fixes the problem. The camera manager should be aware of the current status of the camera and avoid calling other functions if the camera is disabled. Also, I did not check if it removes the prompt for photo/microphone permissions on startup. It does notqualify for a pull request I guess that's why I post it here. (the commit has a lot of changes, my editor removed the trailing whitespaces).

lwansbrough, 2015-07-10:
Just added some changes which should address this issue. Give it a try and let me know if you notice anything. Thanks!

closed 2015-07-10 by lwansbrough