react-native-camera/react-native-camera #54: [torchMode] Cannot read property 'on' of undefined

uoroaix, 2015-06-20:
Hi, I am not sure if this is me or is torchMode currently not working?
Here is my code 
```
var Camera = require('react-native-camera');
var QRCodeScreen = React.createClass({
  getInitialState() {
    return {
      cameraTorch: Camera.constants.TorchMode.on,
      // other states...
    };
  },
 getDefaultProps: function() {
    return {
      cameraTorch: Camera.constants.TorchMode.on,
      // other properties...
    };
  },
})
```
And I create a Camera like this
```
  render: function() {
    return (
      <Camera 
        onBarCodeRead={this._onBarCodeRead} 
        style={styles.camera} 
        type={this.state.cameraType} 
        orientation={this.props.orientationType}
        aspect={this.state.cameraAspect}
        torchMode={this.props.cameraTorch}
      >
        <View style={styles.rectangleContainer}>
          <View style={styles.rectangle}/>
        </View>
        <View style={styles.cancelButton}>
          <TouchableOpacity onPress={this._onPressCancel}>
            <View>
              <Text style={styles.cancelButtonText}>{this.props.cancelButtonTitle}</Text>
            </View>
          </TouchableOpacity>
        </View>
      </Camera>
    );
  },
```
i've also tried torchMode={this.state.cameraTorch}
but it always says Camera.constants.TorchMode is undefined thus the error Cannot read property 'on' of undefined.

Just wondering if I am using this correctly?

lwansbrough, 2015-07-02:
Can you confirm that `Camera.constants` is populated?

lwansbrough, 2015-07-16:
Going to close this until someone can reproduce.

closed 2015-07-16 by lwansbrough

mkhanal, 2016-04-27:
I have the same problem now, just fetched the npm from github right now.
for this code revision : 75dff8bff1d556ba65eba5597d7b2bde0d34a733

mkhanal, 2016-04-27:
Through some debugging I can see that CameraModule is undefined after line 12

const CameraManager = NativeModules.CameraManager || NativeModules.CameraModule;
-> console.log(CameraManager) here shows that CameraManager is undefined