react-native-camera/react-native-camera #56: Add metadata to image.

jtremback, 2015-07-05:
Gonna attempt to add this today.

jtremback, 2015-07-06:
Probably going to do this as an optional parameter that one can pass to `.capture`. Seems kind of weird semantically, but the react native geolocation polyfill is too good not to use.

jtremback, 2015-07-06:
Might be good to actually have that parameter just be general metadata.... or maybe pass it via props?

jtremback, 2015-07-08:
Adding GPS metadata here: https://gist.github.com/jtremback/e5dde527da9bcd69536e

I still need to sort out some merging stuff with the latest version tomorrow. @lwansbrough you may want to bump the version that is on npm to the latest.

lwansbrough, 2015-07-11:
https://github.com/lwansbrough/react-native-camera/commit/8eb6e9e7dce7d70b9be60a70515a7b972f9a8a40

baseio, 2015-07-31:
Hi, sorry for writing to a closed issue - but some additional documentation/examplecode on the "pass a location object from the geolocation api" would be helpful...
Thanks in advance!

lwansbrough, 2015-07-31:
@jtremback? I personally haven't done this.

jtremback, 2015-08-01:
Sure, I'll take care of it.

jtremback, 2015-08-02:
Does this clear things up? 00a2b3adaf3df877da75181e5889664dbd4033b7

closed 2015-08-02 by jtremback

RikNieuwoudt, 2016-06-29:
Hi, I still can't figure out how to add meta data to an image.
If we use the example on the Usage section on the repo homepage:
`  takePicture() {
    this.camera.capture()
      .then((data) => console.log(data))
      .catch(err => console.error(err));
  }`

Where and how do I add the meta data?

npomfret, 2017-03-19:
hi - i've submitted a PR which I think fixes this on iOS (android to follow).

The way to use it is to something like:

  async _takePicture() {
      const options = {};
      if (this.state.position) {
        // position comes from: navigator.geolocation.watchPosition((position) => this.setState({position}))
        options.location = this.state.position;
      }
      const data = await this._camera.capture(options);
  }

Then when you use RN CameraRoll to retrieve the image it has the location info attached.  I'm looking for someone to help test my PR.  If you have any time it would be appreciated.

https://github.com/lwansbrough/react-native-camera/pull/635

koswarabilly, 2017-11-17:
@npomfret have android version been fixed? because i need to pass metadata to images