react-native-camera/react-native-camera #57: Use of UIImage strips some metadata from photo

jtremback, 2015-07-07:
In my quest to add GPS metadata, I have found that converting to and from `UIImage` strips some metadata from an image https://gist.github.com/jtremback/50cf75604c9e5c55db49.

It appears that the only reason that `UIImage` is being used is to satisfy the requirements of the `UIImage+Resize` class that is included in the module. It appears that `UIImage+Resize` is not actually doing anything, just resizing the image to the same size and orientation that it already was. So, eliminating `UIImage+Resize` and the conversion to `UIImage` results in images being saved with full metadata. Here's a very quick and dirty modification that saves images without the conversion: https://gist.github.com/jtremback/b4996f4e884f4ce0748e.

HOWEVER- While `UIImage+Resize` is not doing anything now, as far as I can tell, I think it should be. There are two schools of thought on image orientation. One way is to have the capturing device set the orientation property in the metadata and let the displaying device rotate it properly. However, displaying devices often don't do this right. There are many examples across the internet of this issue. The other way is to rotate the image in the capturing device and leave the orientation metadata untouched. This is what I want my app to do. It's probably also what `UIImage+Resize` is supposed to be doing in the code.

Fixing the call to `UIImage+Resize` to rotate properly will be easy, but preserving the metadata may not be. I can think of two ways to do this:

- Rotate without converting to `UIImage`, possibly with `CGImage`. Perhaps this will be kinder to the metadata.
- Read the metadata, rotate with `UIImage`, and reattach the metadata afterwards. 

I don't know yet which is the best approach, but I will update this tomorrow.

jtremback, 2015-07-08:
Successfully stopped stripping the metadata here: https://gist.github.com/jtremback/e5dde527da9bcd69536e

Still need to do the orientation flip.

Unfortunately I wrote this on an older version, so I will need to sort out the merge tomorrow.

lwansbrough, 2015-07-10:
Hey @jtremback, thanks for your work regarding this. I think we should do it the way it's *supposed* to be working, which is to transpose the X/Y coords in the final image. This is just simpler for everyone and typically represents what people want the camera to do (otherwise they'd have to manage rotation themselves by reading metadata.) 

I think copy/pasting the metadata is a decent solution, so long as it doesn't require doing a ton of remapping. Ideally it's a few lines of code that map every piece of metadata that exists now or in the future.

lwansbrough, 2015-07-11:
https://github.com/lwansbrough/react-native-camera/commit/8eb6e9e7dce7d70b9be60a70515a7b972f9a8a40

closed 2015-07-11 by lwansbrough