react-native-camera/react-native-camera #61: Stuttery viewfinder on switching to camera

jtremback, 2015-07-11:
It's kind of dim and stuttery for about a second. Trying to take a picture during this time will crash the app.

lwansbrough, 2015-07-19:
Not sure we're gonna fix this Jehan, it's pretty difficult to manage the camera in such a way taking into consideration how the views are managed by React Native. I'd support your effort to try, but I did a number of optimizations and they didn't seem to help substantially with this particular issue.

amccloud, 2015-09-01:
@lwansbrough I'd like to try to tackle this. Do you happen to know why it isn't playing nice with react?

I'm observing two behaviors. One for first-open and for re-open.

#### First-open
- Shows first frame of live feed dim (~150ms)
- Switches to another frame of live feed dim (~100ms)
- Switches to live feed no longer dim

#### Re-open
- Shows last frame when the camera was last opened (~150ms)
- Switches to live feed dim (~150ms)
- Switches to live feed no longer dim

dhrrgn, 2015-10-13:
:+1: For getting this fixed.  I would help out, but don't have much experience in this area...

lwansbrough, 2015-10-20:
Please confirm this is still an issue with the latest version of this module + React Native.

amccloud, 2015-10-20:
@lwansbrough 

On `react-native-camera@0.3.8` and `react-native@0.11.0`

#### First-open
- Shows first frame of live feed dim (~200ms)
- Switches to live feed no longer dim

#### Re-open
- Shows last frame from first-open and never switches to live feed
- SIGABRT at the line below when capture is called

```objc
[self.stillImageOutput captureStillImageAsynchronouslyFromConnection:[self.stillImageOutput connectionWithMediaType:AVMediaTypeVideo] completionHandler:^(CMSampleBufferRef imageDataSampleBuffer, NSError *error) {
```

dhrrgn, 2015-10-20:
@lwansbrough  Upgraded to both react-native-camera@0.3.8 and react-native@0.12.0

Same as @amccloud above on first open.  It is better than it was however (only shows 1 dim frame instead of multiple "stutters").

On re-open, I also see the last frame from first open, however it is only for a few milliseconds (maybe 100-200), then it switches to live feed, and I am able to capture another image and/or video.

amccloud, 2015-10-20:
@dhrrgn that's weird the re-open doesn't behave the same. I think i've identified the issue.

https://github.com/lwansbrough/react-native-camera/blob/master/ios/RCTCameraManager.m#L19

Prevents RCTCamera from initializing and starting a new sessions. Setting `self.camera` to nil when the session stops fixes the issue but i'm not sure that is the intention. Could also call `    [self.camera.manager startSession];` in `#RCTCameraManager.view`

amccloud, 2015-10-20:
@dhrrgn @lwansbrough sorry I actually was on 0.3.7 :cry:

Everything works like @dhrrgn described.

closed 2016-06-17 by charpeni