react-native-camera/react-native-camera #62: Long pause between hitting shutter and picture being taken

jtremback, 2015-07-11:
The first picture is taken instantly, however pictures in quick succession will have a long delay before being taken. Hitting the shutter button quickly enough can crash the app.

lwansbrough, 2015-07-11:
Do you think this has something to do with your image rotation code?

jtremback, 2015-07-13:
Possibly. I'll look into it.

achang28, 2015-07-13:
I'm having a similar issue.  The camera is "paused" for about 15 seconds after the Camera component renders.  But after the it works just fine (provided I don't re-render it). One way of getting around this problem is to not unmount the containing component

achang28, 2015-07-13:
I'm having a similar issue.  The camera is "paused" for about 15 seconds after the Camera component renders.  But after the it works just fine (provided I don't re-render it). One way of getting around this problem is to not unmount the containing component. I'm using 0.3

jtremback, 2015-07-13:
@achang28 I think you may be referring to this issue: https://github.com/lwansbrough/react-native-camera/issues/61

What I'm talking about here is something that happens when you hit the shutter several times in quick succession. Tomorrow, I'll try to track down the cause.

lwansbrough, 2015-07-13:
@jtremback If it happens to be the rotation code, maybe using the metadata is best?

jtremback, 2015-07-13:
If it is the rotation code, then it's due to some flaw in the code, not the fact that the image is being rotated. In that case, I'll switch back to the library that you were using before to do the rotation via resize.

jtremback, 2015-07-13:
Yes, it is the rotation code. I'll try some different code. Rotating an image is well within the capabilities of the iPhone.

closed 2015-07-14 by lwansbrough