react-native-camera/react-native-camera #64: What type of bar codes can be read?

nraboy, 2015-07-14:
I'm sorry if this is an inappropriate place to ask, but what types of bar codes can be read with this component?  Maybe you can add them to the README?

Thanks,

jtremback, 2015-07-14:
I've never used the barcode functionality, but I saw this in the source:

```
           @"BarCodeType": @{
               @"upce": AVMetadataObjectTypeUPCECode,
               @"code39": AVMetadataObjectTypeCode39Code,
               @"code39mod43": AVMetadataObjectTypeCode39Mod43Code,
               @"ean13": AVMetadataObjectTypeEAN13Code,
               @"ean8":  AVMetadataObjectTypeEAN8Code,
               @"code93": AVMetadataObjectTypeCode93Code,
               @"code138": AVMetadataObjectTypeCode128Code,
               @"pdf417": AVMetadataObjectTypePDF417Code,
               @"qr": AVMetadataObjectTypeQRCode,
               @"aztec": AVMetadataObjectTypeAztecCode
               },
```

nraboy, 2015-07-14:
Awesome.  Thanks for finding this for me @jtremback :-)

lwansbrough, 2015-07-16:
@nraboy Yeah that's the list, also the scan event will return the type of barcode recognized.

closed 2015-07-16 by lwansbrough

liukepeng, 2015-12-16:
I want to  implement  scan QR_Code with React Native for Android.Can you help me?

lwansbrough, 2015-12-16:
@liukepeng Android is not yet supported unfortunately.

liukepeng, 2015-12-17:
Thank you very much!