react-native-camera/react-native-camera #65: Enhancement Request:  Image Resolution Size

achang28, 2015-07-14:
Would an enhancement request for setting image resolution size make sense?

jtremback, 2015-07-15:
I was thinking of doing this, I have no real need for it myself right now though.

lwansbrough, 2015-07-16:
@achang28 I'd support a PR for this, probably best to do it through the `.capture()` method options. If you're not interested, it's probably something we'll get around to at some point.

achang28, 2015-07-16:
Was thinking of possibly giving it a shot.  If it takes some Obj-C skills && has a bit of complexity to it, then i'm outta luck.  Although I wonder whether my 6 weeks of full-time Swift skills may be useful...

lwansbrough, 2015-07-17:
It'll be objective c but shouldn't be too complicated from what I can recall.

jtremback, 2015-07-17:
@lwansbrough where will it happen? can we get it in the size we want from the AVCapture session, or does it need to be resized?

lwansbrough, 2015-07-17:
@jtremback It happens here: https://github.com/lwansbrough/react-native-camera/blob/e983a6902f2abd5930fc829502867da3c5b63183/RCTCameraManager.m#L101

I think that's what we're talking about? If we're talking about changing the size of the photo based on some given dimensions, I don't think that's actually the job of the camera..

achang28, 2015-07-17:
I think what would be most appropriate (for the Camera) at this point would be to require "quality level" to be provided as (Integer) props, e.g. 1 to 100.  Then this module (react-native-camera) can simply use the provided number as the multiplier for the native camera resolution.

Guessing this would be the simples -- yet decently valuable -- solution.

morukutsu, 2015-07-18:
For me, switching to AVCaptureSessionPresetPhoto did the trick. However, I don't think this is compatible with video.

For reference: https://developer.apple.com/library/prerelease/ios/documentation/AVFoundation/Reference/AVCaptureSession_Class/index.html#//apple_ref/doc/constant_group/Video_Input_Presets

darkrishabh, 2015-07-22:
It is compatible. I just tried something.. I will create a PR tomorrow. but it is not functional. I would like to request you guys to have a look at it and test it thoroughly. I am not good with obj-c but I have a working code that does what we want.

darkrishabh, 2015-07-22:
Better I am putting a zip file of the whole project till then. just unzip and npm start. Don't do npm install.
https://www.dropbox.com/s/xhnzcyvsc7v2pir/cameraApp%202.zip?dl=0

jtremback, 2015-07-27:
@lwansbrough is this something we want to support in this module?

lwansbrough, 2015-07-27:
Not extensively. However, I think we should support the adjustment of the output in the line I mentioned earlier on: https://github.com/lwansbrough/react-native-camera/blob/e983a6902f2abd5930fc829502867da3c5b63183/RCTCameraManager.m#L101

achang28, 2015-11-19:
I believe the solution to this request lies in https://github.com/lwansbrough/react-native-camera/issues/119

closed 2015-11-19 by achang28