react-native-camera/react-native-camera #67: Phone still heats up

darkrishabh, 2015-07-18:
I tried it using for image capture and it significantly increases the phone temperature
And my code properly mounts and un mounts the component. So there is nothing like it is being kept initiated. 

Compared it with the ios camera itself.
Tested on -
iPhone6 [ iOS8.3, 8.4, & 9 ]

morukutsu, 2015-07-18:
Yes, it does. As I pointed out in the previous issue on the subject, the culprit might be there: https://github.com/lwansbrough/react-native-camera/blob/master/RCTCameraManager.m#L147

I guess a little housekeeping (eg. calling stopRunning on the AVCaptureSession instance) is needed before the component is unmounted.

achang28, 2015-07-18:
@darkrishabh: out of curiosity, what exact things are you doing with the Camera module when unmounting?

lwansbrough, 2015-07-19:
@morukutsu Oh yes I may have accidentally left that lingering. There's a React Native issue triage happening tomorrow, maybe I'll get some of these bugs fixed up then too.

lwansbrough, 2015-07-19:
@darkrishabh @morukutsu @achang28 Please give this branch a try: https://github.com/lwansbrough/react-native-camera/tree/performance-improvements

lwansbrough, 2015-07-19:
I'm hoping this is fixed. If everyone could please update to `0.3.1` and report back that would be great, thank you.

achang28, 2015-07-19:
so with 0.3.2, the overheating does NOT appear to be happening. Just to be clear, the sequence I was following that was leading to this problem went like this:  1) Render Camera module, 2) unmount Camera module, 3) press home button on iPhone, and 4) leave iPhone sitting around for 5+ minutes. 

Re-running this sequence and the phone does not seem to be overheating anymore.

lwansbrough, 2015-07-19:
Great news. I'll close this issue then. :)

closed 2015-07-19 by lwansbrough

darkrishabh, 2015-07-21:
@lwansbrough Perfect!! and its seamless.