react-native-camera/react-native-camera #68: Long pause immediately after Camera initializes

achang28, 2015-07-18:
When rendering the react-native-camera module, the app locks up for about 15 seconds before I can interact with the app.  Can't scroll; can't navigate anywhere, etc.  Seems like a case of locking the UI thread.  Anyway, after the 15 seconds, the camera works fine.

lwansbrough, 2015-07-19:
@achang28 Thanks for reporting this, I haven't experienced this but I'm going to look into it tomorrow.

lwansbrough, 2015-07-19:
So I saw and replied to your pull request, waiting for your thoughts. In the mean time, here's something I'm proposing. This fixed UI lock up for me, let me know if it works for you.

https://github.com/lwansbrough/react-native-camera/tree/performance-improvements

achang28, 2015-07-19:
Caused a crash (in Xcode) when I unmounted react-native-camera module.  See attached image.

So I Downloaded everything from "performance-improvements" branch, and restarted everything (Xcode, and  terminal instance running Node (iojs v2.3.4)).
[image]

lwansbrough, 2015-07-19:
@achang28 Try moving line 97 to before the super call on line 95 and report back.

achang28, 2015-07-19:
oh Damn!  Just retried the app about 15x (e.g. relaunched in Xcode).  Works like a charm!  Awesome shit!  I'll keep playing around with it, but it has been working consistently with "only" a 0.5 second delay after rendering the react-native-camera module.  Consistent 0.5 second delay beats 15 seconds any day!!!

lwansbrough, 2015-07-19:
Ok cool, that's good to hear. I think I'll merge this with master then push to npm.

closed 2015-07-19 by lwansbrough

achang28, 2015-07-19:
Merge it!

achang28, 2015-07-19:
looks like your suggested change of ```[self.manager stopSession]``` above ```[super removeFromSuperview]``` didn't make it through w/ 0.3.1.  In RCTCamera.m, around line 95

lwansbrough, 2015-07-19:
Oh sorry, I guess that wasn't clear. This change is needed to fix the problems you mentioned?

achang28, 2015-07-19:
Yeah.  Specifically the app crashing that occurs when unmounting Camera module.

lwansbrough, 2015-07-19:
My bad, `0.3.2` published to accommodate these changes.

achang28, 2015-07-21:
Hot like buttah.  Good shit.  Works beautifully.

lwansbrough, 2015-07-21:
👍

commit 3fb44f27 (2015-07-19): 0.3.1 fixes #68 and #69