react-native-camera/react-native-camera #72: Import module issue

chentsulin, 2015-07-21:
When I run the program it seems that some problems happened with the `require` function：

```
Unable to resolve module React from /Users/chentsulin/Projects/iOSCamera/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module RCTDeviceEventEmitter from /Users/chentsulin/Projects/iOSCamera/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module NativeModules from /Users/chentsulin/Projects/iOSCamera/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module ReactNativeViewAttributes from /Users/chentsulin/Projects/iOSCamera/node_modules/react-native-camera/Camera.ios.js
```
In my previous experiences in browser and node, `require` function should have a module name or relative path as parameters.
How can these to be imported?
```
var React = require('React');
var DeviceEventEmitter = require('RCTDeviceEventEmitter');
var NativeModules = require('NativeModules');
//....
```
Sorry for as a react-native novice, I can't find out how to be done. Any Inspiration with be appreciated.

achang28, 2015-07-21:
@chentsulin:  are you simply testing out the Camera module?  If you are, please remove all modules and be sure to only have:

```var Camera = require("react-native-camera");```
```var React = require("react-native");```

I'm assuming you followed the instructions on adding the necessary files to your Xcode project (e.g. RCTCamera.xcodeproj)?  And making sure that you stopped your Xcode project, and restarted it?

lwansbrough, 2015-07-21:
@chentsulin Here's the `index.ios.js` code for the app I use for testing:

```javascript
/**
 * Sample React Native App
 * https://github.com/facebook/react-native
 */
'use strict';

var React = require('react-native');
var {
  AppRegistry,
  StyleSheet,
  Image,
  Text,
  View,
  TouchableHighlight
} = React;
var Camera = require('react-native-camera');

var rncamera = React.createClass({

  getInitialState: function() {
    return {
      isRendered: true
    };
  },
  
  takePicture() {
    this.refs.cam.capture({}, this.onCapture);
  },

  start() {
    this.refs.cam.capture({
      mode: 'video'
    }, this.onCapture);
  },

  stop() {
    this.refs.cam.stopCapture();
  },
  
  onCapture(err, data) {
    if (err) return console.log(err);
    this.setState({
      lastCapture: data
    });
  },
  
  onBarCodeRead(e) {
    console.log(e);
    this.setState({ barcodeData: JSON.stringify(e) });
  },
  
  toggleRender() {
    this.setState({
      isRendered: !this.state.isRendered
    });
  },

  render: function() {
    
    var camera;
    if (this.state.isRendered) {
      camera = (
        <Camera
          ref="cam"
          captureTarget="cameraRoll"
          type="back"
          onBarCodeRead={this.onBarCodeRead}
          style={styles.camera}
        >
          <TouchableHighlight style={styles.btn} onPress={this.start}><Text>Start Video</Text></TouchableHighlight>
          <TouchableHighlight style={styles.btn} onPress={this.stop}><Text>Stop Video</Text></TouchableHighlight>
          <TouchableHighlight style={styles.btn} onPress={this.takePicture}><Text>Take Picture</Text></TouchableHighlight>
          <Text style={styles.barcodeData}>{this.state.barcodeData}</Text>
        </Camera>
      );
    }
    
    return (
      <View style={styles.container}>
        {camera}
        <Image style={styles.lastCapture} source={{ uri: this.state.lastCapture }}/>
        <TouchableHighlight style={[styles.toggleRender, styles.btn]} onPress={this.toggleRender}><Text>Toggle Render</Text></TouchableHighlight>
      </View>
    );
  }
});

var styles = StyleSheet.create({
  container: {
    flex: 1,
    alignItems: 'stretch',
    backgroundColor: 'transparent',
  },
  camera: {
    flex: 1,
    alignItems: 'center',
    justifyContent: 'center'
  },
  btn: {
    backgroundColor: 'rgba(255,255,255,0.5)',
    padding: 10
  },
  lastCapture: {
    position: 'absolute',
    left: 10,
    bottom: 10,
    height: 60,
    width: 60
  },
  toggleRender: {
    position: 'absolute',
    right: 10,
    bottom: 10
  },
  barcodeData: {
    alignSelf: 'flex-end'
  }
});

AppRegistry.registerComponent('rncamera', () => rncamera);
```

chentsulin, 2015-07-21:
This morning I using npm upgraded my `react-native` and `react-native-camera`, and I thought that break the packager build. I removed node_modules and installed again. Then, it works fine.

But I remain curious about module system in `react-native`. It seems that packager use filenames to address the modules. If I have a naming conflict with react-native core files, will it break packager build?

closed 2015-07-21 by chentsulin