react-native-camera/react-native-camera #73: Image quality

darkrishabh, 2015-07-22:
how can we set image quality as for video, a hd video of 20 sec is over 50mb which is not something you want to upload from your data plan.

darkrishabh, 2015-07-22:
I will be making a PR for that, since its not there and a camera plugin should have this. :)

lwansbrough, 2015-07-22:
Having that discussion here I believe. Noted, I'll try and figure something out ASAP.

lwansbrough, 2015-07-22:
Oops, forgot the link: https://github.com/lwansbrough/react-native-camera/issues/65

darkrishabh, 2015-07-22:
I did try something but what it was pretty unusual. It worked good for the first time but taking a videos back to back raised error during save

BTW I have react-native-cameraIOS repo which uses SCrecorder and give alot of additional features. you can check it out. It provides filter also. I think it can be a great library to help us throughout. I pulled it from react-native screcorder and fixed few things as it was quite inactive there. Have a look at that and also the screcorder library for objective c

lwansbrough, 2015-07-22:
@darkrishabh We've decided not to rely on third party libraries at this time due to the complexity of managing third party dependencies.

jtremback, 2015-07-22:
@darkrishabh SCRecorder looks pretty full featured, perhaps you can just wrap that and use it in your app without involving this module.

darkrishabh, 2015-07-22:
Yeah that what I am doing over here https://github.com/darkrishabh/react-native-cameraIOS

closed 2015-07-27 by jtremback