react-native-camera/react-native-camera #74: Unrecognized tag protocol

slonoed, 2015-07-24:
I use 
`captureTarget={Camera.constants.CaptureTarget.disk}`
When photo captured, callback receive string like
`/var/mobile/Containers/Data/Application/9B63344B-4B73-46AB-B8E8-0C447E001303/Documents/86ABB2D5-9C18-49C5-9DE0-37B35367AE23.jpg`

Then I use this string in Image
`<Image source={{uri: data}}/>`

First issue is this message (but I see the image)
```
2015-07-25 01:56:45.311 [warn][tid:main][RCTStaticImageManager.m:59] Error setting property 'imageTag' of RCTStaticImage with tag #24: Unrecognized tag protocol: /var/mobile/Containers/Data/Application/9B63344B-4B73-46AB-B8E8-0C447E001303/Documents/86ABB2D5-9C18-49C5-9DE0-37B35367AE23.jpg
```

Second issue: the image not shown (I suppose the cache is cleared or smth else).

Should I do smth with image, i.e. move to another directory to save it?

slonoed, 2015-07-24:
For first I found solution. isStatic:true

lwansbrough, 2015-07-25:
@slonoed You probably don't want a static image (though I'm not sure, I forget -- I usually use `cameraRoll`) -- what you want is to prepend the URI with `file://` however if we're returning the protocol on one and not the other that's probably not the best behaviour.

slonoed, 2015-07-25:
@lwansbrough it works fine without `file://`, I don't know why.

My only issue is cache. I think it isn't issue with react-native-camera, but with develop process.

When I restart app with xcode, the path is changed. I'm talking about folder in `Application` folder.

`/var/mobile/Containers/Data/Application/0CE546E7-DA58-42A3-8701-8CFA55456E1C/Documents/1F82781E-0213-4E77-8504-87172EDA0724.jpg`

`/var/mobile/Containers/Data/Application/9B63344B-4B73-46AB-B8E8-0C447E001303/Documents/86ABB2D5-9C18-49C5-9DE0-37B35367AE23.jpg`

I tried to find how change this behavior in xcode but without success.

lwansbrough, 2015-09-15:
@slonoed Yeah hmm, not sure how we can solve this..

closed 2015-11-13 by lwansbrough