react-native-camera/react-native-camera #76: Orientation sometimes fails to initialise correctly

JedWatson, 2015-07-26:
I'm seeing an intermittent error when the camera is initialised, where the orientation is incorrectly detected.

Using an iPad Mini 2 w/ iOS 8.4, starting in Landscape mode, it sometimes starts in Portrait mode. There doesn't seem to be a reliable way to detect the current orientation in React Native itself (?) otherwise I'd set the `orientation` prop, but my app can be used in either `landscapeLeft` or `landscapeRight` (locked in XCode)

Not sure if this is related to my project never allowing portrait orientation, that may mess with the detection?

Otherwise it looks like this may be a timing issue, it works properly about 80% of the time.

If I reorient the device, the camera fixes itself and stays fixed, until the next time the camera is initialised.

jtremback, 2015-07-27:
You're talking about the orientation of the viewfinder, right?

JedWatson, 2015-07-29:
@jtremback yes, that's correct.

lwansbrough, 2015-10-20:
Is this still an issue?

JedWatson, 2015-10-21:
It's still an issue, and actually a pretty major one for us, I just haven't been able to get someone with enough iOS / ObjectiveC experience to look into how to fix it yet :grimacing:

Any chance you could take a look? I can set up a project that replicates the issue if that will help.

lwansbrough, 2015-10-21:
I might have some time, no guarantees though. Pretty busy myself. AFAIK React Native doesn't have great support for horizontal display mode at this point. Haven't checked into that in a while though.

corbt, 2016-02-05:
Any workarounds @JedWatson? I'm running into this as well, which is a blocking issue because our app only records in landscape mode. I'm finding that consistently when the phone is already in landscape mode and I render the `Camera` the preview is in portrait orientation, until I wiggle the device a bit and it snaps to landscape. If I render the `Camera` when the phone is in portrait and then rotate the device, it behaves correctly.

Interestingly, if I set the `orientation` prop to any value except `auto`, including `landscapeLeft`, the preview is locked in the portrait orientation and never changes.

If I figure out a fix or workaround I'll post it here.

### Landscape phone with incorrect portrait preview
[image]

### Landscape phone with correct landscape preview
[image]

corbt, 2016-02-05:
I think I've tracked down the bug to the following method:

```objc
RCT_EXPORT_METHOD(changeOrientation:(NSInteger)orientation) {
  NSLog(@"Ok, manager sees orientation as %ld", (long) orientation);
  self.previewLayer.connection.videoOrientation = orientation;
  NSLog(@"Ok, we've set the videoOrientation as %ld!", (long) self.previewLayer.connection.videoOrientation);
}
```

On the first iteration through, `previewLayer.connection.videoOrientation` *always* remains 0, ignoring what we try to set it to. When the method is called a second or more times (in response to the device being rotated), the `videoOrientation` is set correctly.

I'm assuming this is related to some initialization order issue where `videoOrientation` is not yet ready to accept a new value the first time we set it. I'm far from an Objective-C expert, so if anyone else can figure out why it doesn't work the first time that would be great.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco