react-native-camera/react-native-camera #78: Barcode scanning slow

LeoFidjeland, 2015-08-07:
Hiya!

First of all, thanks for your great work! This is awesome!

I'm having some issues with the Barcode scanning being slow. Sometimes it takes like 15 seconds of the barcode being fully in focus before it gets read.

I put some breakpoints in didOutputMetadataObjects, and it seems like it's Apple code that is the problem.
I'm just wondering if anyone else is having the same issue? And if there is something I can do to speed it up.
I'm comparing with a few other iOS apps out there (like NeoReader) that scan the same code super fast.

LeoFidjeland, 2015-08-08:
Maybe this is still valid...
https://blog.sabintsev.com/apples-native-ios-barcode-scanner-is-junk-83d2129bd017

lwansbrough, 2015-08-08:
It's possible that those faster scanners use a custom library if that's true about the speed of the native barcode scanner. That's unfortunate. We're sticking to Apple's libraries for this module for simplicity. Let me know if you discover anything.

LeoFidjeland, 2015-08-09:
Ok will do. My plan is to see if it works better with QR-codes than with Aztecs. Otherwise make a RN wrapper for a custom library. Currently my eyes are on https://github.com/TheLevelUp/ZXingObjC

LeoFidjeland, 2015-08-09:
Yeay I switched to QR codes and the performance is much better! Seems like it was the Aztec code in particular that was crappy.

Thanks for for attention :)

closed 2015-08-09 by LeoFidjeland

danna88, 2017-05-03:
i have the same issue in android! Can you tell me the way to solve this problem?

zorosun, 2017-08-07:
I also have the same issue with old mobile phone, it seems that the phone is unable to correctly focus on the QR

augusto-altman, 2017-11-04:
I have the same issue on Android!

abdulrahman-khankan, 2017-12-28:
Same thing on Android, trying to read EAN-13 codes but it's too slow. With Expo's API demo app it's super fast and it's built on top of this :/ Not sure if I'm doing something wrong.

MohsenKhosravinia, 2019-12-16:
you can use the below answer:

https://github.com/react-native-community/react-native-camera/issues/1002#issuecomment-566018391