react-native-camera/react-native-camera #80: Camera freezes when opened 2 times via navigator

hrlarsen, 2015-08-26:
Hi

Currently have an app where you are meant to take several photos. Currently I have a class handling the camera capturing (CameraCapture.js), and this is opened using the IOS navigator with a push from a "capture images" class (imagine a number of empty thumbails that needs to be filled by capturing photos). When the CameraCapture is pushed the first time everything is fine and I can capture a picture etc. But when I then try to open it again it freezes. 

The freeze does not happen when I pop the "capture images" class and open it again. So it's like the camera is not properly destroyed or something when the component is unmounted, and only destructs when the class holding the CameraCapture class is unmounted.

hrlarsen, 2015-08-26:
I found that this happens because CameraCapture variable ```javascript var CameraCapture = require('./CameraCapture') ``` creates an instance of the CameraCapture object, and this cannot handle showing the <Camera> more than once.

achang28, 2015-08-26:
I hate to say it, but I, too am still experiencing freezes much like @hrlarsen is experiencing (even though 0.3.3 reduced the original problem). I wonder if there is some sort of "cleanup" or additional mounting process that should be done in the obj-c world, in addition to what has already been built in react-native-camera?

lwansbrough, 2015-09-15:
@hrlarsen Interesting. I'll look more into this sometime soon.

zubru, 2015-10-09:
+1 I also have problem with this. When I go back from view with camera and once again go into it, camera freezes

jforaker, 2015-10-09:
+1 
Does anyone have a solution for this?

jforaker, 2015-10-09:
SOLVED:

http://stackoverflow.com/a/33048580/1516309

@zubru 
@lwansbrough 
@achang28 
@hrlarsen

lwansbrough, 2015-10-10:
Thanks Jakeed1 of StackOverflow for this :) I'll implement sometime tonight hopefully. Tomorrow latest.

jforaker, 2015-10-10:
That is me ;)

lwansbrough, 2015-10-10:
@jforaker Oops hahah, of course. Thanks!

lwansbrough, 2015-10-10:
Can you guys give it a shot? `0.3.7`

closed 2015-10-10 by lwansbrough

gpbl, 2015-10-11:
I'm using 0.3.7 but the camera still freezes as i go back to the view using it

lwansbrough, 2015-10-11:
@gpbl Can you confirm that you completely reinstalled the camera module? It's possible the binaries haven't updated. It's also possible that this fix wasn't the proper fix. I haven't tested it myself.

gpbl, 2015-10-11:
I can confirm, 0.3.7 was the first version I ever installed 😊 I wonder how @jforaker got it fixed.

lwansbrough, 2015-10-11:
I too am interested. Hopefully he'll respond here so we can get more info on this fix.

jforaker, 2015-10-11:
I'll post everything when I get home ..  
Just to confirm- did you 
clean
Stop server
Build 
Refresh
?

jforaker, 2015-10-11:
Well i can guarantee what I have works, as the answer on SO came from my teammate and we are working on the same project :D 

in RCTCameraManager.h
add this
```
@property (nonatomic, strong) RCTCamera *camera;
```
[image]

in RCTCameraManager.m
replace this block:
```
   return [[RCTCamera alloc] initWithManager:self bridge:self.bridge];
```

```
    if(!self.camera){
        self.camera = [[RCTCamera alloc] initWithManager:self bridge:self.bridge];
        return self.camera;
    }
    return self.camera;
```
[image]

Seems to be exactly as @lwansbrough did in the update. Maybe some compiling issue where on the Xcode side for you? 

Here is me clicking back and forth on the camera (please disregard the atrocious layout - also as you know you will not see the image since i'm using the simulator)  
http://jffileshares.s3.amazonaws.com/Screen-Recording-2015-10-11-19-21-27-f5WgI0Z7VL.mp4

gpbl, 2015-10-12:
@jforaker How can you see it is not frozen, since in the simulator it's always black? :-) 
I do have the last version installed and I see with the debugger that `self.camera` is reused once it's already initialized. I'll investigating further...

gpbl, 2015-10-12:
Without knowing what does it mean, if i set `self.camera = nil;` in `RCTCameraManager.m it works (ie it does not freeze).
@lwansbrough does it make sense? 
[image]

jforaker, 2015-10-12:
The problem we had before was that the library was recreating an instance of the view every time which contains many objects in it like the manager etc. Therefore, it spiked up the memory and froze the device. The fix will be to reuse the view instance. @gpbl I am glad it worked. And yes we should probably nill the camera instance when the session is stop.
Also I've been testing on my phone, just dont know how to do a screencast to show you

gpbl, 2015-10-12:
@jforaker it sounds like your problem was not related to the navigator... but yes that issue was not good as well. I trust it works for you :-) (you can take video from your iphone with quicktime, btw)

toddw, 2015-10-12:
@jforaker plug your phone in with USB, open QuickTime and "record new movie". Select iPhone as the source

mpatric, 2015-10-13:
Just to add to this, the fix in this commit: https://github.com/lwansbrough/react-native-camera/commit/8cc61edef2c018b81e1c52f13c7d261fe6a35a63 introduces issues for us too. We were not previously seeing the camera view freezing when returning to the view (and we do use a Navigator), but in 0.3.7 it does now freeze after being displayed the second time. The workaround suggested by @gpbl fixes the issue for us. Having a quick look at the lifecycle of the views, it looks like this is the correct approach.. anyone care to agree or disagree?

jforaker, 2015-10-13:
I agree. nilling `self.camera` was a good idea that we missed the first go around.

lwansbrough, 2015-10-13:
@mpatric @jforaker Seems to be the correct fix yes. I can do that work and push to npm this evening, at work right now. If someone wants to submit a PR for that, I'd be happy to accept it - but I can't get the changes into npm until later today anyway.

commit 8cc61ede (2015-10-10): Fixes #80