react-native-camera/react-native-camera #81: No Location in the MetaData

thomashermant, 2015-08-26:
Hi,

I do not have the location meta data when I take a picture with this Camera Component.

--------------------

I search the reason on the web and found this:
"
The problem is that since iOS 4 UIImage *image = [info objectForKey:UIImagePickerControllerOriginalImage]; strips the geolocation out. To solve this problem you have to use the original photo path to get access to the full image metadata. With something like this:

- (void)imagePickerController:(UIImagePickerController *)picker didFinishPickingMediaWithInfo:(NSDictionary *)info {
    NSURL *referenceURL = [info objectForKey:UIImagePickerControllerReferenceURL];
    ALAssetsLibrary *library = [[ALAssetsLibrary alloc] init];
    [library assetForURL:referenceURL resultBlock:^(ALAsset *asset) {
        ALAssetRepresentation *rep = [asset defaultRepresentation];
        NSDictionary *metadata = rep.metadata;
        NSLog(@"%@", metadata);

        CGImageRef iref = [rep fullScreenImage] ;

        if (iref) {
            self.imageView.image = [UIImage imageWithCGImage:iref];
        }
    } failureBlock:^(NSError *error) {
        // error handling
    }];

"
**from this link**: [http://solvedstack.com/questions/reading-the-gps-data-from-the-image-returned-by-the-camera-in-ios-iphone](http://solvedstack.com/questions/reading-the-gps-data-from-the-image-returned-by-the-camera-in-ios-iphone)

--------------------

I wish to update the code but my problem is that I'm a very new by in objective C and there are to much things I don't understand about the codes.

Could you integrate the modification to resolve the location of the picture?
Best regards,

Thomas.

lwansbrough, 2015-09-15:
Hey @thomashermant thanks for reporting this issue. I'll take some time to make some much-needed upgrades to the camera module later this week. Hope you can hold out 'til then. Cheers.

alubling, 2015-12-05:
Just reading this issue because i'm also trying to get location data. I think this is the working thread quoted above, old but seems to capture the issues and has some more recent suggestions for fixes. http://stackoverflow.com/questions/10302250/reading-the-gps-data-from-the-image-returned-by-the-camera-in-ios-iphone. 

For the time being it seems like an easier solution is simply to grab location data as a separate event at the time the picture or video is taken.

joshuapinter, 2017-03-12:
Did this ever go anywhere? For now I'll try and get the location separately, as @alubling suggests.

npomfret, 2017-03-19:
hi - i've submitted a PR which I think fixes this on iOS (android to follow).

The way to use it is to something like:

  async _takePicture() {
      const options = {};
      if (this.state.position) {
        // position comes from: navigator.geolocation.watchPosition((position) => this.setState({position}))
        options.location = this.state.position;
      }
      const data = await this._camera.capture(options);
  }

Then when you use RN CameraRoll to retrieve the image it has the location info attached.  I'm looking for someone to help test my PR.  If you have any time it would be appreciated.

https://github.com/lwansbrough/react-native-camera/pull/635

koswarabilly, 2017-11-17:
@npomfret
i have this code in my project

```
takePicture() {
    const options = {};
    options.location = this.state.location;
    this.camera.capture({ metadata: options })
      .then((data) => console.log(data))
      .catch(err => console.error(err));
  }
ever
```ytime i press capture the application stop working

ospfranco, 2018-01-12:
@npomfret already submitted PR and was already fixed. closing this.

closed 2018-01-12 by ospfranco