react-native-camera/react-native-camera #83: Camera Flash doesnt seem to work

viscasillas, 2015-09-01:
```
var React = require('react-native');
var Icon = require('react-native-vector-icons/FontAwesome');
var {
  AppRegistry,
  StyleSheet,
  Text,
  View,
  TouchableHighlight
} = React;
var Camera = require('react-native-camera');

var cameraApp = React.createClass({
  getInitialState() {
    return {
      cameraType: Camera.constants.Type.back,
      cameraFlashToggle: Camera.constants.FlashMode.on
    }
  },

  render() {

    return (
      <Camera
        ref="cam"
        style={styles.container}
        onBarCodeRead={this._onBarCodeRead}
        type={this.state.cameraType}
        flashMode={Camera.constants.FlashMode.on}
      >
        <View style={styles.buttonContainer}>
          <TouchableHighlight onPress={this._takePicture} style={styles.snapButton}>
            <Icon name="camera-retro" size={55} color="#6AC9C8" style={styles.snapIcon} />
          </TouchableHighlight>
          <TouchableHighlight onPress={this._switchCamera} style={styles.switchButton}>
            <Icon name="circle-o-notch" size={15} color="#ffffff" style={styles.switchIcon} />
          </TouchableHighlight>
          <TouchableHighlight onPress={this._flashEnabled} style={styles.flashToggleButton}>
            <Icon name="bolt" size={15} color="#ffffff" style={styles.flashToggleIcon} />
          </TouchableHighlight>
        </View>
      </Camera>
    );
  },
  _onBarCodeRead(e) {
    console.log(e);
  },
  _flashEnabled(){
    var state = this.state;
    state.cameraFlashToggle = state.cameraFlashToggle === Camera.constants.FlashMode.on
      ? Camera.constants.FlashMode.off : Camera.constants.FlashMode.off;
    this.setState(state);
  },
  _switchCamera() {
    var state = this.state;
    state.cameraType = state.cameraType === Camera.constants.Type.back
      ? Camera.constants.Type.front : Camera.constants.Type.back;
    this.setState(state);
  },
  _takePicture() {
    this.refs.cam.capture(function(err, data) {
      console.log(err, data);
    });
  }
});

var styles = StyleSheet.create({
  container: {
    flex: 1,
    justifyContent: 'center',
    alignItems: 'center',
    backgroundColor: 'transparent',
  },
  buttonContainer:{
    flex:1,
    marginTop:420,
    width:150,
    height:150,
    marginLeft:55
  },
  snapButton:{
    width:100,
    height:100,
    backgroundColor:'transparent',
    borderRadius:50,
    alignItems:'center',
    justifyContent: 'center',
    borderWidth: 5,
    borderColor: '#6AC9C8'
  },
  snapIcon:{
    width:50,
    height:50
  },
  switchButton:{
    width:30,
    height:30,
    backgroundColor:'transparent',
    borderRadius:15,
    alignItems:'center',
    justifyContent: 'center',
    borderWidth: 2,
    borderColor: '#ffffff',
    marginLeft: 80,
    marginTop: -10
  },
  switchIcon:{
    width:15,
    height:15
  },
  flashToggleButton:{
    width:30,
    height:30,
    backgroundColor:'transparent',
    borderRadius:15,
    alignItems:'center',
    justifyContent: 'center',
    borderWidth: 2,
    borderColor: '#ffffff',
    marginLeft: -10,
    marginTop: -30,
    paddingLeft:7
  },
  flashToggleIcon:{
    width:15,
    height:15
  },
});

module.exports = cameraApp;
```

viscasillas, 2015-09-01:
I have tried using the state for cameraFlashToggle = {this.state.cameraFlashToggle} and I cant seem to swap that one either

Jackarr, 2015-09-04:
Please help me

lwansbrough, 2015-09-15:
@viscasillas Your problem seems to be here:

```javascript
_flashEnabled(){
    var state = this.state;
    state.cameraFlashToggle = state.cameraFlashToggle === Camera.constants.FlashMode.on
      ? Camera.constants.FlashMode.off : Camera.constants.FlashMode.off;
    this.setState(state);
  },
```

The ternary operator returns false in either case, so your flash mode will always be false.

closed 2015-09-15 by lwansbrough