react-native-camera/react-native-camera #84: unable to resolve module react 0.10.1

spyro92, 2015-09-05:
is anyone else getting these errors in the react packager - react 0.10.1

ive followed all the instructions in the readme

```
Unable to resolve module React from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module RCTDeviceEventEmitter from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module NativeModules from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module ReactNativeViewAttributes from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module StyleSheet from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module createReactNativeComponentClass from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module ReactPropTypes from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module StyleSheetPropType from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module NativeMethodsMixin from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module flattenStyle from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module merge from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module React from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module RCTDeviceEventEmitter from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module NativeModules from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module ReactNativeViewAttributes from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module StyleSheet from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module createReactNativeComponentClass from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module ReactPropTypes from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module StyleSheetPropType from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module NativeMethodsMixin from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module flattenStyle from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
Unable to resolve module merge from /Users/nicksqy/dev/nom/node_modules/react-native-camera/Camera.ios.js
```

philkeys, 2015-09-05:
Yeah getting the same issue on build.

DeodorantMan, 2015-09-06:
Same thing

PR #86 (2015-09-15): fix imports and add merge as a dependency
these changes restore compatibility with react native 0.10 and resolve #84
closed 2015-09-15 by lwansbrough