react-native-camera/react-native-camera #87: Android support

ardalann, 2015-09-16:
Hey guys, fantastic job! Thank you!

I was just wondering whether you have any plans to add Android support as well.

achang28, 2015-09-20:
+1,001

lwansbrough, 2015-09-20:
Hey guys, I'm working with a few folks from @exponentjs on the next iteration of rem which will support Android builds. Currently I'm not using rem but I think it is a good thing for the community going forward, so I'm investing some time in helping with that effort. FB currently doesn't have a recommended approach to Android module development - we're trying to solve that. Ideally this will allow you to install your modules like `npm install react-native-camera` and they will automatically be added to your XCode projects and Android dependencies and built with your app. Until then I can try to get a PoC for the Android camera but it'll take some time to get feature parity with the iOS version.

achang28, 2015-09-21:
@lwansbrough:  do you accept donations?  Glad you're championing the "camera space" of React Native

lwansbrough, 2015-09-21:
@achang28 Thanks for considering donating! However I'm not doing this for the money, I've gone a personal interest in keeping the development of this up to date. Always looking for contributors though -- I think more of the community modules could use a larger backing of contributors to keep them fresh.

lwansbrough, 2015-09-22:
Small update: moved the iOS files around in preparation for Android dev. :+1:

ardalann, 2015-09-22:
Thank you very much @lwansbrough ! :thumbsup:

mattiamanzati, 2015-09-22:
Thanks a lor! I Will wait For Android Version!

jiamo, 2015-09-22:
https://www.npmjs.com/search?q=react-native-camera version is 0.3.3 . How can I use the lastest version in my project?

lwansbrough, 2015-09-22:
:grimacing: Oops forgot to publish! Try now.

timmh, 2015-09-23:
I gave it a shot and got the camera view working on android: https://github.com/timmh/react-native-camera
It is very primitive and simple but it took some time to figure out how to include the native component into the app, so someone may find it useful.

lwansbrough, 2015-09-23:
Nice work, this is definitely something we can build on. I will mention that if it's anything like Obj-C, you will need to inherit React's view class so that you get all their view stuff like flexbox, etc.

timmh, 2015-09-23:
I think the whole style & layout works via `super.updateView(view, props);` in the ViewManager.
Just tried different layouts including flexbox and the sizing worked as it should.

tlightsky, 2015-10-06:
hi, @timmh how can i use your android version?

timmh, 2015-10-06:
@tlightsky there is an README.md in the [android directory](https://github.com/timmh/react-native-camera/tree/master/android) that has instructions how to use it.
@lwansbrough I am currently implementing barcode scanning using the zxing library, would you be interested in merging this in?

mattiamanzati, 2015-10-06:
@timmh I am also interested in it! Hoping that you Will merge!

lwansbrough, 2015-10-06:
We need a way of depending on 3rd party libs first. Currently we have to include the binary that we're targeting which sucks.

lwansbrough, 2015-10-06:
With that said, since I don't think Android has bar code scanning by default (correct my if I'm wrong) then yes I'd be interested in using zxing. Historically it has been the position of this library to not depend on third party libraries but that may need to be the case for Android feature parity.

What I really want is a build script that links to a Maven repo though. But we need to standardize something like that across React Native before doing it here.

timmh, 2015-10-06:
You are right, unfortunately android does not have barcode scanning included.
A possible solution would be to create two different gradle configurations which compile the library with and without 3rd party barcode scanning libraries so the user could choose whether he needs this feature or not. But it seems you are working on something that makes this kind of thing easier, so I understand that you want to wait until that is ready. Already curious about that!

JorisM, 2015-10-08:
@timmh thanks for your work! i pulled and tried to get it to work, but I ran into issues here:
first there is a typo in the readme in the android directory, I think it should look like this:

`project(':com.lwansbrough.ReactCamera').projectDir = new File(settingsDir, '../node_modules/react-native-camera/android')`

(maybe it would also be worth mentioning that the lines for build.gradle are meant to be added in the android/app/build.gradle, not android/build.gradle, I got confused with that until I figured it out.)

second, after I got everything to compile, and I start the app, it instantely crashes with this error: 

```
E/AndroidRuntime(21835): java.lang.AssertionError: Module RCTNetworking was already registered!
E/AndroidRuntime(21835): 	at com.facebook.infer.annotation.Assertions.assertCondition(Assertions.java:65)
E/AndroidRuntime(21835): 	at com.facebook.react.bridge.NativeModuleRegistry$Builder.add(NativeModuleRegistry.java:162)
E/AndroidRuntime(21835): 	at com.facebook.react.ReactInstanceManager.processPackage(ReactInstanceManager.java:447)
E/AndroidRuntime(21835): 	at com.facebook.react.ReactInstanceManager.createReactContext(ReactInstanceManager.java:416)
E/AndroidRuntime(21835): 	at com.facebook.react.ReactInstanceManager.recreateReactContext(ReactInstanceManager.java:345)
E/AndroidRuntime(21835): 	at com.facebook.react.ReactInstanceManager.onJSBundleLoadedFromServer(ReactInstanceManager.java:310)
E/AndroidRuntime(21835): 	at com.facebook.react.ReactInstanceManager.access$100(ReactInstanceManager.java:70)
E/AndroidRuntime(21835): 	at com.facebook.react.ReactInstanceManager$1.onJSBundleLoadedFromServer(ReactInstanceManager.java:97)
E/AndroidRuntime(21835): 	at com.facebook.react.devsupport.DevSupportManager$13$1.run(DevSupportManager.java:530)
E/AndroidRuntime(21835): 	at android.os.Handler.handleCallback(Handler.java:739)
E/AndroidRuntime(21835): 	at android.os.Handler.dispatchMessage(Handler.java:95)
E/AndroidRuntime(21835): 	at android.os.Looper.loop(Looper.java:135)
E/AndroidRuntime(21835): 	at android.app.ActivityThread.main(ActivityThread.java:5274)
E/AndroidRuntime(21835): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(21835): 	at java.lang.reflect.Method.invoke(Method.java:372)
E/AndroidRuntime(21835): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:909)
E/AndroidRuntime(21835): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:704)
```

any idea why this might happen/how to fix it?

again, thanks for your work!

JorisM, 2015-10-08:
ah btw, the crash happens after adding this line: 

`.addPackage(myReactPackage)`

to the MainActivity.java

timmh, 2015-10-08:
Hey @JorisM, thanks for your feedback!
I am sorry that the instructions in the readme are not working for you.
Did you replace the existing `.addPackage(whateverPackage)` with `.addPackage(myReactPackage)`?
If not, that might be the issue.
I will look into this with more detail later today and update the instructions.
However, I would ask you and anyone else having an issue with my fork to create an issue over there because we should not bloat this issue with problems not directly associated with it.

JorisM, 2015-10-08:
ah yes, that was it! thanks a bunch for reporting back so quickly!

yup I agree about not spamming this thread, but I couldn't create an issue on your fork since issues don't seem to be enabled.

ortutay, 2015-10-08:
in @timmh's version, I get the following error when I do `this.refs.cam.capture`:

```Error: this.refs.cam.capture is not a function```

When I investigate the Camera component in the debugger, the `capture` method is indeed not there. Any ideas what's going on?

timmh, 2015-10-08:
@JorisM You are right, I did not notice that issues were disabled. Happy that you now seem to have it working.
@ortutay my version is just a concept  / placeholder until the stable and feature-complete version is released in *this* repo. As the [readme](https://github.com/timmh/react-native-camera/blob/master/android/README.md) states, my version is just an experiment and is neither stable nor feature-complete (e.g. it hast no capture-method as you already noticed).

Issues are now enabled for my fork so if anyone has questions / issues regarding that, please fill an issue [there](https://github.com/timmh/react-native-camera/issues).

tlightsky, 2015-10-13:
I can run the camera on Andorid with @timmh 's version.
I notice the component is quite different from iOS version.
Will they be kind the same at the end?

joshhornby, 2015-10-25:
@lwansbrough Is there an ETA on this?

lwansbrough, 2015-11-12:
Yeah guys if you not +1 this endlessly. I understand this is in high demand and don't have the time to implement it right now. Going forward I will have a need to get the camera working on Android which is probably when it will end up happening. No timeline yet. If you've come up with a solution you think is suitable please don't hesitate to submit a PR, even if it isn't feature-complete.

lifuzu, 2015-11-26:
Did anyone try this: https://developers.google.com/vision/barcodes-overview.
It seems Google support Barcode scanner within Play Services, here is the sample project: https://github.com/googlesamples/android-vision

dangnelson, 2015-12-11:
Might be worth it to mention up front in the docs that this is iOS only. I made the dummy mistake of assuming this was cross platform since it wasn't explicitly mentioned, and it's for react-native...

lwansbrough, 2015-12-11:
@dangnelson Sorry about that Dan! It will eventually.. I can definitely add that to the docs.

dangnelson, 2015-12-11:
many thanks, otherwise great module 🙌

chirag04, 2016-01-07:
@lwansbrough I got https://github.com/journeyapps/zxing-android-embedded working with react-native. I can PR if you would like to add it to react-native-camera.  

Android barcode scan needs to be trigged manually tho. So we can add scan as a method on CameraView.

```js
Camera.scan((args) => {
      console.log(args);
    });
```

thoughts?

lwansbrough, 2016-01-09:
Should be an instance method to coincide with the current API, but yes I'd be okay with using that library I suppose. Does that mean we'd pull in their code into this repo? Or how were you thinking of doing it? We still don't have a way to do this :(

jhogendorn, 2016-01-11:
I assume you've all spotted this? https://github.com/NorthFoxz/react-native-camera-android

Any thoughts?

lwansbrough, 2016-01-11:
I just opened an issue to ask if @NorthFoxz wants to merge the modules. Hopefully he's interested! :) Nice find @jhogendorn

rt2zz, 2016-01-12:
@lwansbrough @NorthFoxz @jhogendorn @timmh 

It looks like there are a few threads going on independently right now. Here is my current understanding of the situation:

[react-native-camera-android](https://github.com/NorthFoxz/react-native-camera-android) has barcode support & disk support
[timmh react-native-camera-android](https://github.com/timmh/react-native-camera) has memory & disk support but callbacks are data first with no error handling

I do not care which base implementation we go with, but I would like to standardize functionality and get this merged back asap. For now I will file some PR's against timmh's fork. Reconciling should be simple with all necessary code pretty much going in the onPictureTaken handler: 
https://github.com/timmh/react-native-camera/blob/master/android/src/main/java/com/lwansbrough/ReactCamera/ReactCameraModule.java#L77
https://github.com/NorthFoxz/react-native-camera-android/blob/4014ab270a334787e58edfbd4eb381a47847f0f6/android/src/main/java/com/ReactCamera/RNCameraInstanceView.java#L202

Thoughts?

lwansbrough, 2016-01-12:
@rt2zz Of course, I forgot about @timmh's fork! Ideally we go with the more stable solution. I don't have any preference over what get's implemented. Standardization is a must. If certain branches are handling things better than others, lets take the best of each where possible. APIs need to be consistent with iOS. If this is an impossibility lets talk about what needs to be changed so we can standardize them.

Is anyone willing to oversee this merging process? I'm happy to help whenever possible, but we kind of need an owner for the Android side of it.

rt2zz, 2016-01-12:
ideally someone with more android experience than myself, but I am happy to reconcile as needed.

lwansbrough, 2016-01-12:
@rt2zz As am I when I have the time. Looking for input from @timmh and @northfoxz regarding how they think it should be done.

I suppose we need to start by looking at which implementation is further along or more stable. Closer feature parity is probably more important than stability at this point, as long as the approach is correct. We should also discuss any necessary/ideal API changes that should happen when we merge this. Do we want to keep callbacks, for example, or do we want to switch to events?

NorthFoxz, 2016-01-13:
There are several issues left in my implementation now:
1. Orientation of the camera is only for portrait now
2. Component views of <Camera /> object
3. Autofocus of the view

I have to fix these issues before merging to iOS support.

rt2zz, 2016-01-13:
Sounds good. I like callbacks.

I forgot about video. Reaching feature parity may take a while but afaik everything is possible (namely passive barcode scan and touch to focus)

rt2zz, 2016-01-16:
@NorthFoxz fyi we are getting decent progress over at https://github.com/timmh/react-native-camera/pull/19

Your project has barcode which is a huge +, it might be good to coordinate efforts moving forward. What do you think?

NorthFoxz, 2016-01-16:
@rt2zz sure i would love to help!
a bit busy lately with my daytime work,
i will update my module and apis this weekend and see if you guys think its ok to merge here

lwansbrough, 2016-01-17:
@NorthFoxz Sounds good - you should send those PRs to @timmh's branch so we don't have to coordinate the merge of Android stuff here. Hopefully that will make bringing Android over as a whole easier.

graywolf336, 2016-01-21:
Looking forward to android support being merged!

dangnelson, 2016-01-21:
Same here, can't wait! Thanks for all the great work on this everybody.

farmisen, 2016-01-21:
I pushed a PR (https://github.com/lwansbrough/react-native-camera/pull/157) from my fork a couple of days ago and forgot to mention it here. I haven't touched java for years and I am definitively not an Android expert so at the very least I expect my code to not be very idiomatic and I might have done some  mistakes that should be pretty obvious to spot for a more seasoned Android developer. 

So far only capturing stills has been implemented, adding video should not be too difficult but I hadn't time yet to look into it. I haven't yet looked much into bar code scanning but that might help http://android-developers.blogspot.com/2015/08/barcode-detection-in-google-play.html. A few other smaller things are in my todo list such as taking in account the orientation props and fixing the improper saved image orientation when using the front camera.

It has only been tested on a nexus 5 and with the genymotion emulator because of its easy camera support and I am looking forward to see how it will behave on other devices.

farmisen, 2016-01-21:
I also realize I might have duplicated effort. I have been working on it over the last couple of weeks and didn't realize some were doing the same. Well, I guess there are now 3 different implementation to consider/cherry pick.

rt2zz, 2016-01-22:
haha @farmisen yeah no worries. I think you might actually have the highest api coverage of any of the forks. I think whoever implements video first will take the cake though.

NorthFoxz, 2016-01-22:
@farmisen nice job!

i think i wont have the time necessary to finish the api merge,

please continue your work and i will help if theres time

dangnelson, 2016-01-22:
Could Android support be without Video until it's ready? Or would that cause too many issues?

farmisen, 2016-01-22:
no hard promise but looking to the android dev doc it does not look like too difficult to implement so I am going to give it a shot today.

lwansbrough, 2016-01-23:
#157 has been merged. All efforts for Android should be directed towards the merged implementation. For those of you with forks, please take a look at what missing and help implement those features if you already have them developed. Thanks so much guys. Tons of great work here.

rohitgoyal, 2016-02-22:
@lwansbrough Is the barcode issue is also sorted out?

rt2zz, 2016-02-22:
@rohitgoyal no. There are two possibilities: either using zxing barcode scanner or using android native barcode scanner. The downside of the latter is that it lacks support for older versions of android.

If you have some time, there is a working android barcode implementation at https://github.com/NorthFoxz/react-native-camera-android that can be used as a reference for implementing here. PR appreciated!

cubbuk, 2016-03-09:
Thanks for the great work =) Currently android version does not support video recording, is there any plan for implementing video feature for android too @lwansbrough  @farmisen?

vespakoen, 2016-04-15:
My initial stab at it here:

https://github.com/vespakoen/react-native-camera/commit/3d0ee9475166d83d9bdb73dfb4cf3ba681e3dd48

I feel I have tried anything, but cannot get a playable video file (there is data in there for sure, it just doesn't play, not on android, not in VLC)

This might be due to my device(s), the Samsung Galaxy S4 Mini, and a Nexus 7.

I commented a lot of the code to hopefully get other people going on trying it out / play with it / report back with findings / results.

To try out my branch, run:

```shell
npm install --save git+https://github.com/vespakoen/react-native-camera.git#master
```

Helpful links:

http://developer.android.com/guide/topics/media/camera.html#capture-video
http://developer.android.com/reference/android/media/MediaRecorder.html
http://developer.android.com/reference/android/media/CamcorderProfile.html
https://android.googlesource.com/platform/development/+/4b737b6/samples/browseable/MediaRecorder/src/com.example.android.mediarecorder/MainActivity.java

marcejohnson, 2016-04-16:
Some more testing options... here is my pull request for code I've been working on for awhile to support recording video on Android:
https://github.com/lwansbrough/react-native-camera/pull/262

- Supports same parameters and values as IOS, including captureMode, captureTarget, captureQuality, etc.
- Updated Readme.md
- Example code

Goal is "drop in" for compatibility with IOS... to make Android and IOS equivalent. I will continue to enhance this as bugs or additional changes are needed since I need this for my own project.  Please send me you feedback, thanks.

wakeless, 2016-08-02:
As for installation (I might not have understood some of what was going on above) it's worth mentioning the `rnpm` https://github.com/rnpm/rnpm project. I've recently used it as an installer for a library and it worked perfectly.

marcejohnson, 2016-08-29:
The feature is officially merged and now part of the repo:
#262

You can close this.

closed 2016-08-30 by ardalann