react-native-camera/react-native-camera #89: Camera crashes in Simulator

bfwg, 2015-09-18:
This issue is same as #71 
[image]
[image]

jtremback, 2015-09-18:
Is it really the same issue as #71? I fixed that a long time ago. #71 involved the camera crashing when you take a picture. Is this still the case for you?

closed 2015-09-19 by bfwg

bfwg, 2015-09-19:
@lwansbrough 
No, It was when I open the project it will stop and show this error, somehow this problem went away. But in the simulator I can't open my camera, I will ask me weather I allow the app to access the camera, and after I click yes, I show nothing but a black background.