react-native-camera/react-native-camera #96: Doesn't work with react-native-webpack-server

sslash, 2015-10-02:
I'm trying to use this with a project based on [https://github.com/mjohnston/react-native-webpack-server](react-native-webpack-server), but I'm getting this error:

```ERROR in ./src/containers/Create/index.js
Module not found: Error: Cannot resolve module 'react-native-camera' in /Users/micgunnu/code/sap/routes/Routes-native/src/containers/Create
 @ ./src/containers/Create/index.js 48:13-43```

lwansbrough, 2015-10-04:
Can't say without a full stack trace or source code. The camera module works though so it's either a problem with the webpack server or your own code.

sslash, 2015-10-04:
Think it has got something to do with the webpack server. Anyway, I went back to normal rn transformer for now, thanks :)

closed 2015-10-04 by lwansbrough

pewallin, 2015-10-13:
@sslash - this is probably due to your Webpack configuration not transpiling react-native-camera in node_modules. A build step for this module would fix this. Or just add it to your webpack.config.js...

thachp, 2015-12-30:
I am experiencing this problem.  How do you "add it to your webpack.config.js" ?