react-native-camera/react-native-camera #98: stop scan method?

yedeba, 2015-10-05:
Hi, stop scan barcode method exists?
Thanks

lwansbrough, 2015-10-06:
Unbind the event handler.

closed 2016-06-17 by charpeni

GarimaMathur07, 2017-09-22:
@lwansbrough We receive multiple (~10x) same barcode in a single camera capture:  using bind and unbind listner approach reduces it for some time but it call again in 2-3 seconds. can you please tell me using code snippet

danilovdev, 2017-11-03:
How to do that? 

> Unbind the event handler.

yedeba, 2017-11-03:
Hi, I simplified it with simple var control.

const intoscan = false;
_onBarCodeRead(e) {
    if(intoscan==false){
        intoscan=true;
        ....

efstathiosntonas, 2018-01-11:
@lwansbrough can you please share some code on how to do that?

jgcmarins, 2018-01-11:
@efstathiosntonas pull requests are always welcome \o/