react-native-camera/react-native-camera #99: Issue with requiring unknown module 'react-native-camera'

josiahbgrace, 2015-10-07:
I'm trying to use the react-native-camera in a project and I keep running into this error:

```
Requiring unknown module "react-native-camera". If you are sure the module is there, try restarting the packager.
```

I've tried everything from a clean build and run, and clean installs of the package. Any help would be appreciated. I'm using version 0.3.6 with react-native version 0.11.4.

viscasillas, 2015-10-07:
have you tried restarting Xcode or the packager itself?

josiahbgrace, 2015-10-07:
Yes, I've tried both of those.

lwansbrough, 2015-10-07:
I haven't looked into this yet so it could be a problem with the module (though I'd doubt they'd break something in a minor version change.) Maybe try clearing your Babel cache? (`rm ~/.babel.json`)

josiahbgrace, 2015-10-07:
Hm... Yeah, that didn't fix the issue either. I'm working on other parts of the project now, but I'll circle back to this later and see if I can do some more digging to add some more concrete details to this issue.

prathamesh-sonpatki, 2015-10-13:
I am also facing same issue with version 0.3.7 with react-native version 0.7.1.

morenoh149, 2015-10-13:
experiencing the same on `0.3.7` react-native `0.11.0`

lwansbrough, 2015-10-20:
@prathamesh-sonpatki Not supporting that many versions back of RN.

@morenoh149 Can you update the camera and verify that this is an issue still?

amccloud, 2015-10-20:
@josiahbgrace @prathamesh-sonpatki  @morenoh149 are you using react-native-webpack-server?

morenoh149, 2015-10-20:
@amccloud yes actually. `"react-native-webpack-server": "^0.7.1",`

amccloud, 2015-10-20:
@morenoh149 It doesn't handle index.ios.js and index.android.js like you'd expect. You can import this way:

```js
import Camera from 'react-native-camera/index.ios';
```

amccloud, 2015-10-20:
@lwansbrough You could probably close this as it is likely a `react-native-webpack-server` issue https://github.com/mjohnston/react-native-webpack-server/issues/96

josiahbgrace, 2015-10-21:
I am not using the react-native-webpack-server, but the code suggested above 

    import Camera from 'react-native-camera/index.ios';

fixes the issue (with the example code in the readme).

closed 2015-10-21 by lwansbrough

blacksunset, 2015-10-22:
@josiahbgrace 
```
import Camera from 'react-native-camera/index.ios';
```
i tried the code above but i got
SystaxError:Unexpected use of reserved word 'import'

josiahbgrace, 2015-10-22:
I don't know what I was thinking when I copied that over.... My actual code is 

    var Camera = require('react-native-camera/index.ios');

https://github.com/josiahbgrace/ClothesMatcher/blob/master/CameraUtil.js

And checkout this file for base code.

blacksunset, 2015-10-22:
@josiahbgrace Thank you, i have fixed the problem.
When i installed the  react-native-camera with command like this
```
npm  install react-native-camera@latest --save
```
the folder “react-native-camera” was created in mode_modules. i just copied the folder into /path/[project_name]/node_modules/  
then completed the steps from 2 - 6 in the readme