react-native-camera/react-native-camera #101: Mirror image on front camera

skeie, 2015-10-10:
Hi,

When picture is taken with the front camera, the image returned is mirrored. Is it possible to return the images 'as-is'? In other words, not mirrored?

lwansbrough, 2015-10-10:
That's default behaviour of the camera. That functionality isn't currently supported but I'd accept a pull request!

dhrrgn, 2016-02-24:
@lwansbrough @skeie It turns out we needed this feature, so... I submitted a PR for it :) (it is only for iOS  at the moment, see PR for reasoning): https://github.com/lwansbrough/react-native-camera/pull/212

gaguirre, 2016-09-08:
Hi guys, we needed this feature for Android too so we implemented it.
We already submitted a PR #399

brunolemos, 2017-08-08:
+1

Tried both combinations of `mirrorImage` but front camera pictures are always mirrored.

gitlovenotwar, 2017-08-30:
I had the same issue, mirrorImage is being ignore. Front camera always return mirrored image.

wassgha, 2017-09-16:
For me `mirrorImage` mirrors the image vertically rather than horizontally...

arminsal1, 2017-09-18:
Any updates? I'm having the same issue.

OrDuan, 2017-10-06:
Same issue here

joecaraccio, 2017-10-15:
any progress on this?

LarsWalter, 2017-10-16:
If someone is having trouble on Android with mirrorImage mirroring vertically rather than horizontally, you can try changing react-native-camera/android/src/main/java/com/lwansbrough/RCTCamera/MutableImage.java line 42:
`m.preScale(-1, 1);`
to
`m.preScale(1, -1);`

Why this is the case I'm not sure. I'm guessing this has to do with what orientation you're taking a picture in, landscape or portrait.

Looking at iOS, I just added this little snippet into captureStill in RCTCameraManager line 644. Probably a better way to this other than through UIImage, but I know very little objective C, so this is what I ended up with.

```
  if (self.mirrorImage) {
    UIImage *tmpImg = [UIImage imageWithData: imageData];
    UIImage *flippedImage = [UIImage imageWithCGImage:tmpImg.CGImage
                                                scale:tmpImg.scale
                                          orientation:UIImageOrientationLeftMirrored];
    imageData = UIImageJPEGRepresentation(flippedImage, 1.0);
  }
```

If it flips the wrong direction, like those having issues with android, change out UIImageOrientationLeftMirrored with one of the others: https://developer.apple.com/documentation/uikit/uiimageorientation?language=objc

How one would fix this for video with front cam, I have no idea.

joecaraccio, 2017-10-16:
yeah, this is great.

Trying to figure out how to fix it for video too however..

unicornRainbows, 2017-12-13:
Yeah mirrorImage flips it vertically, the fix in the MutableImage.java would work... but dont like hacking node_modules.. as we all know we going to have to npm install a few times in the RN lifecycle.. but hey thats why we do this shit.. cause normal coding is lame! 

"If you not living on the edge, you taking up too much space!"

unicornRainbows, 2017-12-13:
This is the answer to fix android BUG or FEATURE.. Sometime i dont know if androids bugs are features... ;)

```<Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={
           { width: Metrics.screenWidth,
            height: Metrics.screenHeight,
            position:'absolute',
             top:0,
             left:0,
            backgroundColor: Colors.transparent}}
          aspect={Camera.constants.Aspect.fill}
type={"front"}
mirrorImage={true}
fixOrientation={true}
          captureTarget={Camera.constants.CaptureTarget.disk}
></Camera>```

Using both
**mirrorImage={true}
fixOrientation={true}**

will give you the desired effect!

Happy coding!

OrDuan, 2017-12-13:
@unicornRainbows first it is deprecated.
second, the docs state:

> WARNING: It uses a significant amount of memory and my cause your application to crash if the device cannot provide enough RAM to perform the rotation. 

I hope the core dev will solve this issue internally without hack around, it's been here for more than 2 years :|

facuescobar, 2017-12-15:
Did anyone solve this issue without using **fixOrientation={true}**, this works to me but as OrDuan said in last comment, it's deprecated and also has an important Warning message

LarsWalter, 2017-12-15:
Seeing how this has gone unaddressed for 2 years, If you need this to be fixed I'd suggest you simply work around it yourself. I ended up writing a native module that rotates the image the way I needed using UIImage as described above for iOS and Bitmap + Matrix for Android. Writing something like that should be fairly straight forward for most I think. I just pass the image returned from the camera to my own module before I do anything else with it. It's not efficient, since now there's 2 separate image operations, but it does the job. I had to base64 encode the images anyways, so I just did it there.

An alternative to writing a module like this would be to fork the camera repo and do the changes I described above and just have your project sit on that fork.

As far as video goes, nobody has come up with anything as far as I can tell. The fixOrientation prop sounds like it only fixes images, but I haven't tried it.

monolithed, 2017-12-19:
@LarsWalter Could you publish that package?

monolithed, 2017-12-19:
```
mirrorImage={true}
fixOrientation={true}
```

This fix is not working for iPhone X!

closed 2018-01-12 by ospfranco

chtrinh, 2018-02-23:
For those trying to deal with images, this  as  style for your view works:
```
transform: [{ scaleX: -1 }]
```

sibelius, 2018-02-23:
RNCamera has this prop as well

Duya3fithou, 2020-03-09:
Now, I have not found a solution at all.Because RNCamera no longer uses the `mirrorImage` and `fixOrientation` properties Someone please help me, thanks..

enaluz, 2020-05-01:
Flip the camera via styling. Just pass `transform: [{rotateY: '180deg'}]` to this style prop. My example:

```
      <RNCamera
        style={{ transform: [{rotateY: '180deg'}] }}
        type={RNCamera.Constants.Type.front}
        { ..props }
      />
```

Happy coding!

timegambit, 2020-10-08:
Styling the camera is not a real solution because the picture is never mirrored.

Fix by adding the following code to line 752 of RNCamera.m:

```
if ([options[@"mirrorImage"] boolValue]) {
        [connection setVideoMirrored:true];
    }
```
Then take a mirrored image by calling takePictureAsync({mirrorImage: true})

juliendevleeschauwer, 2021-07-05:
options = { quality: 1, exif: true, writeExif: true, mirrorImage: true, orientation: "portraitUpsideDown"}

DB-009, 2021-07-28:
> options = { mirrorImage: true, orientation: "portraitUpsideDown"}

that didnt work

juliendevleeschauwer, 2021-07-28:
> > options = { mirrorImage: true, orientation: "portraitUpsideDown"}
> 
> that didnt work
with this code my project works well 
cameraType : is a variable containing the type of RNCamera (front back)

const options = {
      mirrorImage: cameraType === RNCamera.Constants.Type.front ? true : false,
      orientation:
        cameraType === RNCamera.Constants.Type.front
          ? RNCamera.Constants.Orientation.portraitUpsideDown
          : RNCamera.Constants.Orientation.auto,
    };
    const cdata = await camera.current?.takePictureAsync(options);

siddharth-kt, 2022-03-30:
Hi @xRv9Fk0oQ6Da, 
any update ?

xRv9Fk0oQ6Da, 2022-04-30:
> Hi @xRv9Fk0oQ6Da , 
> 
> any update ?
Nope. I just sis the above method. Will continue my research when my Project resumes.

DevSubhashis, 2022-05-26:
The ultimate Solution is for Andrloid

    `const options = { 
                quality: 0.5, 
                base64: true,
                mirrorImage: cameraType === RNCamera.Constants.Type.front ? true : false,
                orientation:
                    cameraType === RNCamera.Constants.Type.front
                    ? RNCamera.Constants.Orientation.portrait
                    : RNCamera.Constants.Orientation.auto,
             };
           
            const data = await cameraRef.current.takePictureAsync(options);`

xRv9Fk0oQ6Da, 2022-05-26:
@DevSubhashis, have you tested the perspective, captured or selfie modes yet? Or does it still need Debugging?

siddharth-kt, 2022-05-26:
Hi, @DevSubhashis @xRv9Fk0oQ6Da, in my case it didn't worked.
I needed to set {orientation: "portraitUpsideDown"} on selfie mode(front camera) to work correctly.

xRv9Fk0oQ6Da, 2022-05-26:
> Hi, @DevSubhashis @xRv9Fk0oQ6Da, in my case it didn't worked.
> 
> I needed to set {orientation: "portraitUpsideDown"} on selfie mode(front camera) to work correctly.

You removed his accidental `` from the code, right?

DevSubhashis, 2022-05-26:
> > Hi, @DevSubhashis @xRv9Fk0oQ6Da, in my case it didn't worked.
> > I needed to set {orientation: "portraitUpsideDown"} on selfie mode(front camera) to work correctly.
> 
> You removed his accidental `` from the code, right?

yes, that's for code formatting in text editor also I only found solution for image, still trying for video

siddharth-kt, 2022-05-26:
Yeah its working for me but with this code for front camera.
With this code. (My orientation is different (not-portrait))
{
    mirrorImage: true,
    orientation: 'portraitUpsideDown',
    base64: false,
  };

xRv9Fk0oQ6Da, 2022-05-26:
> > > Hi, @DevSubhashis @xRv9Fk0oQ6Da, in my case it didn't worked.
> 
> > > I needed to set {orientation: "portraitUpsideDown"} on selfie mode(front camera) to work correctly.
> 
> > 
> 
> > You removed his accidental `` from the code, right?
> 
> 
> 
> yes, its working on my phone, that's for code formatting in text editor also I only found solution for image, still trying for video 

If I remember right when I did mine like four years ago, I think I had to mirror the Video. But nothing else, just the View/Modal.

siddharth-kt, 2022-05-26:
@DevSubhashis which android device did you tested on ? mine is vivo.
Our orientation in code is not same, different code is work for you and different is working for me so this might cause issues in production, that's why i am asking.
(look at this comment : https://github.com/react-native-camera/react-native-camera/issues/101#issuecomment-888032711)

also @xRv9Fk0oQ6Da what do you think ?

DevSubhashis, 2022-05-26:
> If I remember right when I did mine like four years ago, I think I had to mirror the Video. But nothing else, just the View/Modal.

yes, for video the dirty solution is, add `transform: [{ scaleX: -1 }]` style to video component

DevSubhashis, 2022-05-26:
> @DevSubhashis which android device did you tested on ? mine is vivo. Our orientation in code is not same, different code is working for you and different is working for me so this might cause issues in production, that's why i am asking. (look at this comment : [#101 (comment)](https://github.com/react-native-camera/react-native-camera/issues/101#issuecomment-888032711))
> 
> also @xRv9Fk0oQ6Da what do you think ?

for my phone its working, its realme 3i with android 10 version, i do't have any idea, if that will work on other devices.

siddharth-kt, 2022-05-26:
@DevSubhashis kindly check it  (if its because of base64 or not 🤔) and let me know.

xRv9Fk0oQ6Da, 2022-05-27:
I think I finally came up with the solution here... Comment #101 has a pin-point concept; props @juliendevleeschauwer, in the issue each Developer is encountering an orientation difference all of which seem to be located in different parts of the world. It seems as if maybe the processor (intel), or other chips which may have their languages transversed could be related to an orientational difference. 

This isn't legit code, but a mere example to help understand what seems to be happening;

`import { NativeModules, Platform } from 'react-native';

const deviceLanguage =
          Platform.OS === 'ios'
            ? NativeModules.SettingsManager.settings.AppleLocale ||
              NativeModules.SettingsManager.settings.AppleLanguages[0] // iOS 13
            : NativeModules.I18nManager.localeIdentifier;

console.log(deviceLanguage); //en_US`

Users in Russia would contain devices with examples such as `en_RU`, India as such `en_IN` and further. 

Let me know how this works out as the solution for you all. 

For most accurate, please consult a Browser View Model while debugging (developing your app) because you're coding for cross-platform, not only one device.

siddharth-kt, 2022-05-27:
@xRv9Fk0oQ6Da yes https://github.com/react-native-camera/react-native-camera/issues/101#issuecomment-888032711
this comment solved my issue.