react-native-camera/react-native-camera #103: Device heats up

nashpro, 2015-10-13:
Hi, thanks for great component.

Is there a way to turn off the camera during transfer to another route? While working in the background, the device gets too hot.

Thanks.

lwansbrough, 2015-10-13:
I'm going to implement a fix later today that stops the camera session when it's removed from its containing view. Perhaps the solution for you then would be to remove the camera from the view when it's not being viewed.

darkrishabh, 2015-10-13:
@nashpro 
you can use the navigator didFocus listeners to activate and deactivate the camera.

On moving to the next route you can change the state and hide the camera, and on coming back use the didFocus listener to bring it back.

lwansbrough, 2015-10-20:
Can you verify that this is still an issue with the latest version?

nashpro, 2015-10-26:
iphone is still heated, we used willFocus to fix this problem.

closed 2017-04-26 by ptomasroos

liubin915249126, 2020-07-28:
> iphone is still heated, we used willFocus to fix this problem.

i wanna to know how to use willFocus to fix this problem. tks a lot.