react-native-camera/react-native-camera #107: Tap to Focus Crashes App

sgonyea, 2015-10-19:
I'm currently trying to debug the source of the issue.  It's dying in a thread, during the execution of `focusAtThePoint:(CGPoint) atPoint` with:
```
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[0]'
*** First throw call stack:
```

It seems to die at some random place within the `dispatch_async` execution.  I'm debugging to see what the source of the issue is.

I'm using React Native 0.11.4 on iOS 9.0.2, with the RN Camera 0.3.8.

Thanks!

sgonyea, 2015-10-19:
Ah, interesting.  This only seems to happen when I have the `onFocusChanged` property defined.  Digging some more...

sgonyea, 2015-10-20:
Seems to be because `self.reactTag` is null, so we're sending an event to React with a target of null here:

https://github.com/lwansbrough/react-native-camera/blob/master/ios/RCTCamera.m#L170-L176

Which is odd.  I'm not yet well versed in React Native's ObjC layers to easily fix this.

amccloud, 2015-10-20:
Reproducible with `react-native-camera@latest` and `react-native@0.11.0`

lwansbrough, 2015-10-20:
@amccloud Thanks!

machard, 2015-10-27:
Same with react native 0.13.2 and onZoomChanged

morukutsu, 2015-11-02:
Hello, can someone test my PR https://github.com/lwansbrough/react-native-camera/pull/115/ and confirm it works on your side? Thanks

sgonyea, 2015-11-10:
@morukutsu: 👍. This fix works for me. Great work! (/cc @lwansbrough)

lwansbrough, 2015-11-13:
@morukutsu I can't pull this in until we can confirm it doesn't affect changes made for #80.

morukutsu, 2015-11-13:
@lwansbrough Yes, I provided a test to repro #80 in my pull request topic. I could not reproduce it with the patch: so either I'm not reproducing the right bug or the bug is simply not there anymore.

Anyway, I tried to ping those who had the bug so they can quickly review my test to see if it's the correct repro but no answers so far :/

realjamie, 2015-11-17:
@morukutsu , your fix works on my project and solves the crashing problem, thanks.

realjamie, 2015-11-17:
When I say the fix from @machard works, it means that the app doesn't crash when touching-to-focus and pinching-to-zoom. The zooming feature works well, but the re-focusing feature doesn't work the same as the iOS camera app does. When you have bright and dark parts in your camera screen at the same time, the brightness is auto-adjusted when re-focusing between the bright and the dark parts. However, this component doesn't do brightness auto-adjustment when the device remains in the same position. Did you implement the component in this way intentionally, @lwansbrough ?

realjamie, 2015-11-18:
Hi @morukutsu, today I tried your fix on iPad and my app still crashes when pinching-to-zoom (but not touching-to-focus). It works well on iPod Touch.

morukutsu, 2015-11-18:
@realjamie Unfortunately, I don't have an iPad to test. I don't use the pinch to zoom feature even on iPhone. 

Do you have additional details (call stack, js error)?
I'll have a look.

realjamie, 2015-11-18:
@morukutsu , thanks for quick reply. Here is the error log:

2015-11-17 17:37:54.939 MyProject[614:911509] *** Terminating app due to uncaught exception 'NSRangeException', reason: '*** -[AVCaptureVideoDevice setVideoZoomFactor:] - the passed videoZoomFactor 1.045100 out-of-range [1, activeFormat.videoMaxZoomFactor]'
*** First throw call stack:
(0x271e5fef 0x358a3c8b 0x25ce4317 0x4d323 0x4f0d9 0x2a9e001b 0x2a88f0a9 0x2aca8e3b 0x2a858131 0x2a85600f 0x2a88d489 0x2a88cd8d 0x2a862f95 0x2aad98ab 0x2a8619a9 0x271abfaf 0x271ab3bf 0x271a9a25 0x270f6201 0x270f6013 0x2eab5201 0x2a8c2a09 0x3f835 0x35e55aaf)
libc++abi.dylib: terminating with uncaught exception of type NSException
(lldb)

morukutsu, 2015-11-18:
@realjamie What is the value of activeFormat.videoMaxZoomFactor?
I guess it's 1.0 on the iPad. Or it's probably related to the current activeFormat.

Apple docs says: "A maximum factor of 1.0 indicates that the format is not capable of zooming."

https://developer.apple.com/library/prerelease/ios/documentation/AVFoundation/Reference/AVCaptureDeviceFormat_Class/index.html#//apple_ref/occ/instp/AVCaptureDeviceFormat/videoMaxZoomFactor

pjcabrera, 2015-11-25:
@realjamie @morukutsu I can fix the crash in iPad if I change line 719 of RCTCameraManager.m to this

`device.videoZoomFactor = zoomFactor >= 1 && zoomFactor <= device.activeFormat.videoMaxZoomFactor ? zoomFactor : 1.0f;`

closed 2015-12-08 by lwansbrough