react-native-camera/react-native-camera #111: popN cause the camera frozen

bfwg, 2015-10-28:
after pop 2 and re-enter camera component it become frozen, I had to exit and re-enter again to make it work again.  
```javascript
_onBarCodeRead(e) {
    if(!this.state.barCode) { // prevent from keep scanning
      this.setState({
        barCode: true,
      });
      this.props.navigator.push({
        title: "AfterScan",
        component: Chat,
        leftButtonTitle: 'Custom Left',
        onLeftButtonPress: () => this.props.navigator.popN(2), /
        passProps: {data: e, onLeave: this._onLeave, token: this.props.token},
      });
      console.log(e);
    }
  },```

mikeljames, 2015-11-25:
same issue for me aswell +1

ethul, 2015-12-11:
Is there a work-around for this by chance? I am getting a similar issue, but it happens when I do `navigator.pop()` and then go back into the route with the camera component. Curious if there are any leads to fixing this.

stantoncbradley, 2016-06-05:
same issue with react-native-navigator-flux, except Actions.pop() gives me error 'NSGenericException', reason: '*** AVCaptureSession can't stopRunning between calls to beginConfiguration / commitConfiguration' so I can't even exit

bjacog, 2016-10-14:
Same issue here, I have two views that utilize the Camera component. The first time either view is presented it works fine when the second view is presented the camera appears sluggish and then grinds to a halt. Diving into the code today to see if we can contribute a fix.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco