react-native-camera/react-native-camera #112: Any way to check if the (iOS) device has flash or front camera?

wootwoot1234, 2015-11-01:
Is there any way to check if the iOS device has flash or front camera?

lwansbrough, 2015-11-03:
Does any device not support flash/front camera these days? :/ I agree this needs to be available if it isn't though.

wootwoot1234, 2015-11-03:
I thought the same thing but iPads don't have flash and of course some older (really old) phones.

https://en.wikipedia.org/wiki/List_of_iOS_devices#Features

lwansbrough, 2015-11-03:
Check out how this PR optionally defines constants: https://github.com/lwansbrough/react-native-camera/pull/109

It's probably possible to do the same checks for camera capabilities. I'll take a look at this at some point, but it's not priority. If you want to try it out and find a nice solution, feel free to open a PR. Thanks!

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco