react-native-camera/react-native-camera #117: Return a dictionary instead of a string in capture()

imothee, 2015-11-05:
I was wondering if it was within scope of the project to propose a change to the interface returned by capture to be a dictionary instead of a string so that you could return more data, somewhat like

https://github.com/marcshilling/react-native-image-picker

What I'd like to do is both write the image to the CameraRoll and return the base64 version of the data within having to jump through multiple hoops. Having a dictionary also allows passing of metadata back from the capture image.

I'd be happy to create a pull-request if this is something that you would be amicable to including. If not, what would you recommend as the simplest way to both persist the image and return base64?

lwansbrough, 2015-11-05:
I suppose I'm curious why you're interested in returning both? Not that I'm against returning an object, but perhaps there is a simpler solution, since you're the only person who's brought up this issue so far.

imothee, 2015-11-05:
I'm trying to avoid using the JS bridge too much. The use case is showing a preview image using the uri (instead of passing in the base64) but having the base64 to start the upload immediately from the JS side.

I realise we could do something like returning the base64, save it to cameraRoll and displaying the image using base64 from the js but that's a triple transfer over the bridge.

We could also write to the asset-library but I found it difficult to access the base64 of the file directly or you have to use cameraRoll which only lets you return {last: 1} and check if it's the same image in case for some reason (background sync, iphoto) it's not and I'm not sure I like relying on an interface you can't specifically get a file by key.

lwansbrough, 2015-11-05:
Personally, I'd probably use something less pretty to upload the image - something like this: https://github.com/kamilkp/react-native-file-transfer But it hasn't been updated in a while. That way you can upload from the Objective C side, removing the need to pass the image over the bridge. With that said, if you think passing an object back would be better I'd probably support that change. Would have to wait for a major version bump before that can be merged in though.

imothee, 2015-11-09:
The upload API takes the file as a base64 string inside a json object, it's simpler to do it with the fetch API, I haven't noticed any performance issues so far. I'll make a fork/pr with an implementation and let you know how it goes.

lwansbrough, 2015-11-10:
> I haven't noticed any performance issues so far.

Passing an image over the bridge will lock the UI. Great, thanks!

closed 2017-04-26 by ptomasroos